dresden-elektronik/deconz-rest-plugin #2617: device support: iCasa switch long-press

nicoduck, 2020-03-22:
In reference to the closed issue: #1124:

The current implementation does not support long-press events on the scene buttons of the switch:
I have a Pulse 8S (ICZB-KPD18S), pressing S1 short logs:
button 3002 Recall scene 1
but long pressing it only logs:
no button handler for: ICZB-KPD18S ep: 0x01 cl: 0x0005 cmd: 0x04 pl[0]: 0x00
The switch only seems to send event when long-pressing but not when releasing (contrary to the on/off buttons)

I'm running deconz in a container, phoscon version 2.05.75 (seems to be latest), with a conbee II USB stick,

What information can I provide to get the long-press events supported?

ebaauw, 2020-03-22:
I need to double-check, but I think the 4S and 8S send a _Store Scene_ command when holding the _S#_ buttons for an extended time.  This is quite longer than your typical long press, so I didn’t expose these.  I would be quite easy to map these commands to x003 button events, but the user experience will be odd, especially as the _On_ and _Off_ buttons react to hold in a normal time frame.

ebaauw, 2020-03-22:
Indeed, _Scenes_ command 0x04 is _Store Scene_.  Looks like the keypads send this when holding the _S#_ button for ~3 seconds.

nicoduck, 2020-03-22:
Thanks for your prompt reply.
The hold time is quite long indeed but should still be okay for some use cases.
I use the switch with home-assistant to start automations and scenes. My plan is to use long presses to deactivate a scene (e.g. cinema mode, which will switch everything off)

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix