dresden-elektronik/deconz-rest-plugin #2618: IKEA Trådfri Remote – buttons only sending x002 events

samuellinde, 2020-03-22:
My IKEA Trådfri Remote (E1810) only sends x002 events on buttons 4 and 5 now, which I'm pretty sure they didn't do a while back. According to the wiki page, they're supposed to support x001 and x003 as well, which they currently don't. I'm listening to deconz_events in the Home Assistant event log, which seems to confirm my findings.

ebaauw, 2020-03-22:
That’s the newer model, isn’t it?  I never laid my hands on that one.  The Wiki is probably about the older model, E1524.  You might check the deCONZ log to see whether the E1810 actually sends something on hold/long release.

mickejo, 2020-03-25:
I can confirm this behavior on the new version of the ikea remote.

jan666, 2020-03-26:
I just tested my IKEA Trådfri Remote-Collection and I think its not the model but the firmware.

I have one E1810 on the old firmware (1.2.223), press and release is triggered:

```
{"e":"changed","id":"10","r":"sensors","state":{"buttonevent":5001,"lastupdated":"2020-03-26T08:57:36"},"t":"event","uniqueid":"14:b4:57:ff:fe:4e:3e:b1-01-1000"}
{"e":"changed","id":"10","r":"sensors","state":{"buttonevent":5003,"lastupdated":"2020-03-26T08:57:37"},"t":"event","uniqueid":"14:b4:57:ff:fe:4e:3e:b1-01-1000"}
```

I have a few E1810 and a few E1524 all on the new firmware (2.3.014), they only send 4/5002

```
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2020-03-26T09:01:37"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":5002,"lastupdated":"2020-03-26T09:01:38"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2020-03-26T09:01:39"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":5002,"lastupdated":"2020-03-26T09:01:40"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
```

ebaauw, 2020-03-26:
Just double-checked: my two E1524 remotes with firmware 2.3.014 do send 4001/4003 and 5001/5003 on long press.

jan666, 2020-03-26:
Hu?

Its impossible to add a video here, so just a photo of my remote:

[image]

```
{
  "config": {
    "alert": "none",
    "battery": 47,
    "group": "13",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "978731b5ee9833a471d9d084df3cec58",
  "manufacturername": "IKEA of Sweden",
  "mode": 3,
  "modelid": "TRADFRI remote control",
  "name": "FB Büro",
  "state": {
    "buttonevent": 5002,
    "lastupdated": "2020-03-26T11:46:19"
  },
  "swversion": "2.3.014",
  "type": "ZHASwitch",
  "uniqueid": "00:0b:57:ff:fe:95:34:36-01-1000"
}
```

```
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2020-03-26T11:46:15"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":5002,"lastupdated":"2020-03-26T11:46:17"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":5002,"lastupdated":"2020-03-26T11:46:19"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
{"e":"changed","id":"17","r":"sensors","state":{"buttonevent":5002,"lastupdated":"2020-03-26T11:46:19"},"t":"event","uniqueid":"00:0b:57:ff:fe:95:34:36-01-1000"}
```

I did "hold and release"

jan666, 2020-03-26:
The other one, E1810 on firmware 1.2.223:

[image]

```
{
  "config": {
    "alert": "none",
    "battery": 74,
    "group": "37606",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "edfbabc9292920ef8aae1323d6d1f3fd",
  "manufacturername": "IKEA of Sweden",
  "mode": 3,
  "modelid": "TRADFRI remote control",
  "name": "FB Küche",
  "state": {
    "buttonevent": 5003,
    "lastupdated": "2020-03-26T12:05:12"
  },
  "swversion": "1.2.223",
  "type": "ZHASwitch",
  "uniqueid": "14:b4:57:ff:fe:4e:3e:b1-01-1000"
}
```

```
{"e":"changed","id":"10","r":"sensors","state":{"buttonevent":4001,"lastupdated":"2020-03-26T12:05:09"},"t":"event","uniqueid":"14:b4:57:ff:fe:4e:3e:b1-01-1000"}
{"e":"changed","id":"10","r":"sensors","state":{"buttonevent":4003,"lastupdated":"2020-03-26T12:05:09"},"t":"event","uniqueid":"14:b4:57:ff:fe:4e:3e:b1-01-1000"}
{"e":"changed","id":"10","r":"sensors","state":{"buttonevent":5001,"lastupdated":"2020-03-26T12:05:11"},"t":"event","uniqueid":"14:b4:57:ff:fe:4e:3e:b1-01-1000"}
{"e":"changed","id":"10","r":"sensors","state":{"buttonevent":5003,"lastupdated":"2020-03-26T12:05:12"},"t":"event","uniqueid":"14:b4:57:ff:fe:4e:3e:b1-01-1000"}
```

djashjones, 2020-03-28:
I can confirm this on the newer version. no hold press xxx1

Old version 2.3.104 1,2 1,2,3 1,2,3 2 2 ( the <> buttons on holds the repeats the events.
Old version 1.2.233 as stated in docs

Wildbill-Z, 2020-04-27:
Strange effect. I have a newer remote control (with the small pairing button) which behaved as described above. Long/short press with the up/down buttons, but only short presses with the <>. Since it only showed 0% battery even with a new one I removed the battery for a while. The I put battery back in, repaired it with Phoscon/deconz (even it existed) and now its showing long/short presses also with the arrow keys. I wonder what happened here.

Greetings, Jürgen

Wildbill-Z, 2020-05-10:
Even stranger effect with another new remote. Showed same behavior as other new remotes, no short/long presses with <> but repeating events. I just removed the battery for a few seconds, pressed some buttons to be sure, no more voltage inside and the put battery back. And without doing something else, I have the functionality of short and long presse even on the <>-buttons.

Greetings, Jürgen

samuellinde, 2020-05-17:
> Even stranger effect with another new remote. Showed same behavior as other new remotes, no short/long presses with <> but repeating events. I just removed the battery for a few seconds, pressed [...]

Thanks for suggesting removing the battery. I just bought a new E1810 which came with the 2.3.014 FW installed. After initial pairing, it didn't register long/short presses. I removed the battery, waited 10 seconds, put it back in and lo and behold – I got long/short presses.

If anyone still has issues with their remotes, I suggest trying this.

closed 2020-05-17 by samuellinde

drreality41, 2020-06-23:
> Even stranger effect with another new remote. Showed same behavior as other new remotes, no short/long presses with <> but repeating events. I just removed the battery for a few seconds, pressed [...]

I can confirm the same behavior and removing the battery and replacing it has solved my problem as well :)

tommi0013, 2020-06-25:
Tradfri remote E1524, firmware 2.3.014

After a lot of tests I now can tell you, how to switch between these two different behaviours without removing the battery.

If your Tradfri remote only shows event 4002-4002, first HOLD on/off button. When event changes from 1002 to 4003, then quickly RELEASE on/off button and DOUBLECLICK < or >. When event changes to 4002 and directly to 1001 (this has to be the last shown event), you successfully (mostly, but not ever) changed to 4001-4003. Maybe you need a few tries. Removing the battery will be faster ;-) 

To switch back from 4001-4003 to 4002-4002, simply HOLD on/off for a few seconds. When event changes from 1002 to 1001, the < and > button will again only show 4002-4002.

Don't know if it is a bug or a feature, but sometimes it can be useful if the tradfri remote sends the 4002 state permanently like the hue dimmer does with the x001 state.

Can anyone confirm this? 

Greetings,
Tommi

ebaauw, 2020-06-26:
> Can anyone confirm this?

Indeed, nice find!

Morale: you don't want to be holding the _On/Off_ button.  I've always thought is was a mistake to support the 1001 for that button...

> Don't know if it is a bug or a feature

Can't answer that either.  It's a fact of life with the current Trådfri remote firmware.  As there's no clue as to what mode the remote is in, and there's no (known) possibility to change the mode over Zigbee, there's nothing we can do in deCONZ.

> sometimes it can be useful if the tradfri remote sends the 4002 state permanently like the hue dimmer does with the x001 state.

Yes, when controlling a device that supports relative change, but no start or stop (like the volume on a Sonos zone player).  This is why I implemented the `hueDimmerRepeat` setting in Homebridge Hue, to map repeating x001 events into repeating _Short Press_ actions in HomeKit.

jloewe, 2020-07-15:
Hi @tommi0013,

> If your Tradfri remote only shows event 4002-4002, first HOLD on/off button. When event changes from 1002 to 4003, then quickly RELEASE on/off button and DOUBLECLICK < or >. When event changes to [...]

I tried it with my E1810 and it works fine! Thank you very much. This is a pretty weird behaviour.

Greetings,
jloewe