dresden-elektronik/deconz-rest-plugin #2619: Problems pairing Philips Smart Button

thorus90, 2020-03-23:
Hi,

I can't get to pair my Philips Smart Button.

What have I done(combinations of these):
- Pressing 3 seconds on the smart button / Pressing the reset button in the back of the device
- Searching for Switches(Add new switch -> Other) / Searching for Sensors (Add new sensor -> Other)

I tried it multiple times, my Philips Dimmer Switch worked on the first time.

I'am starting deCONZ with the systemd service provided by the apt repository. I also tried to stop the service and start the GUI (with X-Forwarding):

[image]

I believe one of these three is the button.
According to #2077 there should be support for this button, right?

Thanks any help is appreciated

ebaauw, 2020-03-23:
> I believe one of these three is the button.

Probably 0x5EF7.  The right circle is missing, meaning deCONZ hasn’t read the descriptors.

> Pressing the reset button in the back of the device

You need to hold the reset for like 10 seconds to reset the button fully.

thorus90, 2020-03-23:
> You need to hold the reset for like 10 seconds to reset the button fully.
I tried 10 sec, 30sec, Holding it the full 3 mins while searching for switches.

So this is basically the correct way to pair the button and it should work?

ebaauw, 2020-03-23:
Basically, yes.  You are on v2.05.75, aren't you?  Note that the button might not yet be supported by Phoscon.

thorus90, 2020-03-23:
> Basically, yes. You are on v2.05.75, aren't you? 

Yes:
[image]
I even compiled the plugin myself on the pi 2.
> Note that the button might not yet be supported by Phoscon.

What do you mean by that? That it is possible it does not show up in the phoscon web app? Is there a way with curl to list all available devices, or for example with home automation, which I do not yet have figured out, how the integration works and how to do stuff there?

ebaauw, 2020-03-23:
Phoscon, the old web app, and home automation plugins/integrations all use the REST API to communicate with deCONZ.  After support for a new device has been added to the REST API plugin, the API clients might need to be updated to support that device as well.  Especially for wireless switches, where the REST API doesn't expose what button events it could generate.

You can manually make REST API calls through `curl`.  I prefer to use `ph`, which is bundled with homebridge-hue.  See the link in the README for the REST API documentation.

thorus90, 2020-03-23:
Ok thanks for clarification!

According to #2077 you own a Smart Button too, does it appear in your Phoscon app?

Listing the complete config with the REST API, I can see a lot Phoscon Clients :smile: , and three sensors. The first one("1") is a software daylight/nighttime sensor depending on my gps, right?

The second("2") is cleary the dimmer switch, because it is called "Dimmschalter". But there is a third one. Perhaps that is already the button?
[Gist to API Output](https://gist.github.com/thorus90/c5dbf1f8adb025bf985430c9990f0c7c)

ebaauw, 2020-03-23:
Yes the second one is your dimmer (modelid RWL021) and the third one is your button (modelid ROM001).  You might have reset it, after it was paired successfully, if it doesn’t react (if state.buttonevent doesn’t change).  Simply reset and re-pair it in that case.

You might need to restart your home automation system before it shows newly paired devices.

thorus90, 2020-03-24:
Thank you very much :) I did need to repair, but now the state.buttonevent changes whenever I press the button.

closed 2020-03-24 by thorus90

tatoosh, 2020-07-21:
Sadly the issue is closed but it helped me a lot.
I know have hue smart button connected and can list at sensor.
But how to use the button now?

Thx!

thorus90, 2020-07-21:
Well you need something like homeassistant to do sth other than turn on/off lamps with it

tatoosh, 2020-07-21:
thanks @thorus90, home assistant is running, can you please give me more infos or samples? thanks

thorus90, 2020-07-22:
I am no expert at all, so I can only tell you what I did.

I modified the automations.yaml
```
- alias: 'open door'
  trigger:
    - platform: event
      event_type: deconz_event
      event_data:
        id: "rom001_3"
        event: 1000
  action:
    service: shell_command.summer
```
and configuration.yaml:
```
shell_command:
  summer: /share/summer
```
and `/share/summer` is a shell skript doing what I want, so every time the button is pressed the shell skript is executed, which in my case contacts the api of my doorbird and opens the door.

tatoosh, 2020-07-22:
Okay thanks for sharing the code!
I solved my problem by configure the hue smart button with the hue essential app on android.
Here are only lights controlled, but running a script can be a next value ;)