dresden-elektronik/deconz-rest-plugin #2621: Mueller Light tint white+color LED E27 causes issues with Alexa

copperfield74, 2020-03-25:
Whenever I change the colors of the light to something other than white or red  my Alexa says that here is something wrong. Within the Alexa app the status from the lights show "Device doesn't support the value". I also own a E14 light from the same company that doen's have this problem and I can choose all colors without problems. I've already deleted the light and reconnected it without success. The light is on your compatibility list. Is this a known problem? I also use iobroker.
Thanks for your help.

taragor, 2020-04-02:
I can confirm that. Happens with my Osram Lightify RGBW Flex Strip. The light changes to the desired color, yet Alexa reports the device not functioning correctly.
"Wohnzimmer Balken scheint nicht richtig zu funktionieren" => Translated: "Wohnzimmer Balken (name of the light) doesn't seem to work properly.
Also it takes a lot longer for this response than the usual "OK"

Mimiix, 2020-06-05:
Is this still a issue?

taragor, 2020-06-05:
Yes. I've updated to the recent version of deconz (Docker version). The issue also persists in the Alexa app. Whenever setting a color other then red (the whitetones also work) the app says 'Device malfunction' yet the light changes as expected. This behaviour is the same for all my RGBW Lights (Lightify FLEX RGBW and Lightify E27 Bulb RGBW)

Mimiix, 2020-06-05:
I am not sure if this is deCONZ at faul or Alexa.

Can you provide more details?

taragor, 2020-06-05:
Sure, I'm using the Docker Version 2.05.75 of Deconz with an ConBee II  stick. All my lights are controllable perfectly fine from Phoscon, Hue Essentials and the REST API (I'm using it with openHAB). All of my RGBW Lights are working perfectly well like this as well.
My lights are:
1x
Vendor: LEDVANCE
Model: CLA60 RGBW Z3
Version: 00103101

2x 
Vendor: OSRAM
Model: Flex RGBW
Version: V1.05.10

When I'm controlling the lights via Alexa (either App or Speaker) I get the following behavior on all 3 lights:
Setting Brightness, White tone, On/Off all work perfectly fine.
Setting color to red works some of the time/ other times it has issues.
Setting any other color allways has issues.

By 'Issues' I mean the lights are changing to the right color but either Alexa says (Translated from german) "<Devicename> Doesn't seem to work correctly", or the App just shows the busy animation (3 dots) and after about 2 seconds it shows a ribbon saying "Device malfunction". The tick showing the color stays at the previous color. After about another 2 seconds the tick switches to the new color.

If you need any more details I'll be happy to provide them.

Mimiix, 2020-06-05:
Hi! Thanks for your answer.

Which application handles Alexa? Because deCONZ doesn't communicate with Alexa directly. Is that OpenHAB in this case?

copperfield74, 2020-06-06:
In my case I use IOBroker! And I still have the problem like Taragor

taragor, 2020-06-06:
@mimiix I've connected Alexa using the "Authenticate App" button in phoscon

Mimiix, 2020-06-12:
I didn't know about that. However, i am not sure what is needed to fix this. I've heard this happens with hue and alexa too (friend of mine had it).

taragor, 2020-07-03:
I think we can close this. The Issue is fixed for me with the current beta release

Mimiix, 2020-07-03:
@taragor As you please!

closed 2020-07-03 by Mimiix

copperfield74, 2020-07-03:
It would be great if the thread opener would be asked if the problem still exists before closing. ;-)
It seems that the problem is solved.

Thanks a lot!

Mimiix, 2020-07-03:
Fair enough :)

ABBolle, 2020-07-05:
Had the same issue with a hue e27 light with deconz 2.05.76, but for me with 2.05.78 the issue is gone.