dresden-elektronik/deconz-rest-plugin #2622: Changing zigbee device modes

Maxtropos, 2020-03-26:
Hello,
I would like to buy one of those Zigbee-LED-Controllers (https://de.aliexpress.com/item/32858603964.html?spm=a2g0x.12010612.8148356.1.28e94419fJlAYw) to drive a simple warm white led (just dimming, no CCT or RGB). As the RGB + CCT versions cost the same as the only dimmable ones, the RGBCCT version is the way to go for me.

My problem is, that this controller would expose itself as RGBCCT. This would make it possible to change the light to a RGB color (for example in home assistant), which would turn the ligh off, as RGB will not be connected.

Is it possible to change the color mode of this Zigbee device from RGB CCT to just dimming (or just CCT), so that it gets exposed to (for example Home Assistant) as a simple dimmable light?

Thank you very much,
Max

ebaauw, 2020-03-26:
Technically: yes, but I don’t think it’s feasible.  There’s no provision in the REST API to specify special handling for a single device (as opposed to all devices of the same type).  Such a change would affect all users of the RGBCCT controller, incl. those that do connect a five-channel light strip.  I don’t think we should be whitelisting individual devices (on mac address) in the code.

Maxtropos, 2020-03-28:
So I can't change the Settings of the device directly in deconz? I will try to change them in Hass.io, hopefully that works.

Thanks!