dresden-elektronik/deconz-rest-plugin #2624: Apt does not show version 2.05.75

bachandi, 2020-03-26:
When checking via apt on http://phoscon.de/apt/deconz for buster stable packages I don't see the 2.05.75 version available. 

```
$ apt-cache policy deconz
deconz:
  Installed: 2.05.74-raspbian-buster-stable
  Candidate: 2.05.74-raspbian-buster-stable
  Version table:
 *** 2.05.74-raspbian-buster-stable 500
        500 http://phoscon.de/apt/deconz buster/main armhf Packages
        100 /var/lib/dpkg/status
```

Did the location of the update change?

jan666, 2020-03-26:
It’s confusing. 2.05.75 is a beta version. Latest stable is 2.05.74

bachandi, 2020-03-26:
Ah I see. Thanks.

bachandi, 2020-05-24:
in the change log https://phoscon.de/de/changelog/ there is no mention of it (2.5.75) being still in beta. Will there be a version change when it becomes stable?

There seems to be some change in apt now but it still has the same version?
```
 apt-cache policy deconz
deconz:
  Installed: 2.05.74-raspbian-buster-stable
  Candidate: 2.05.74-raspbian-buster-stable
  Version table:
     2.05.74-raspbian-buster-stable 500
        500 http://phoscon.de/apt/deconz buster/main armhf Packages
 *** 2.05.74-raspbian-buster-stable 100
        100 /var/lib/dpkg/status

```