dresden-elektronik/deconz-rest-plugin #2625: Websocket added notifications

guillaume1388, 2020-03-27:
Hello,

I encounter a problem with added notification from websocket. When I add a sensor, the websocket json contains the json of sensor. But when I add a light, the json of light is not present in the websocket json whereas documentation says it should be.

Is it normal ? How can I have the json light added?

Here the json I received 
{
    "e": "added",
    "id": "3",
    "r": "lights",
    "t": "event",
    "uniqueid": "7c:b0:3e:aa:0a:07:6e:96-03"
}

I use the version 2.5.74 on windows whith light from osram.

Best Regards,

Guillaume Majorel

ebaauw, 2020-03-27:
It's a bug alright.  I have fixed it already in my local fork, but I'm waiting for @manup to review and merge https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2553 before pushing more changes.

guillaume1388, 2020-03-27:
Ok, thank you for quickly responding.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix