dresden-elektronik/deconz-rest-plugin #2627: Log flooded

espen4001, 2020-03-29:
Hi! I found a similar issue, but it is closed, and I am not sure if it is a bug or normal operation based on the responses in that one, so I’ll ask again.

My Deconz log is flooded by the following message:

```
09:41:54:195 no button map for: SMRZB-332 ep: 0x02 cl: 0x0B04 cmd: 0x0A pl[0]: 005,
09:41:54:195 ZCL attribute report 0x0015BC00140301AD for cluster: 0x0B04, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
```
The device is a smart power relay, with power and current readings.

Is this expected behavior, or is it a bug which should be fixed?

Smanar, 2020-03-29:
I think you have same problem than https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2595

SwoopX, 2020-03-29:
Is that a newly paired device? Have you tried restarting deconz?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix