dresden-elektronik/deconz-rest-plugin #2628: Osram Smart+ Motion sensor does not report temperature

tosion, 2020-03-29:
Hi there,
I have couple similar Home Assistant systems on Raspberry 3B+ where I use Deconz addon and Deconz RaspBee shields. They have been very reliable and I am very pleased with them. I have nodes between 40-70 connected to each system. I have had 4 Osram Smart+ Motion sensors and I added one more. I like them and I also use the temperature measurement adjusted with HA.

This last one just does not report the temperature. I removed it and rejoined again and same result. In Phoscon it shows 0 C (should be around 20) and in HA it reports unknown. In VNC it looks for me just as the others. (I am not an expert here). Can anyone please help me out to figure out is this just broken or is there something I can check deeper?

[image]
[image]
[image]

SwoopX, 2020-03-29:
Based on your screenshot, the modelID may have changed. Can you please double check if the "older" sensors are "Motion Sensor-A"? You should do this in deconz GUI and have the basic cluster attributes read before.

tosion, 2020-03-29:
Thanks for a quick response!
OK - I am not sure is this what you want but new one on the top and old below - both looks same to me?

[image]

[image]

SwoopX, 2020-03-29:
Alright, thanks. Just wanted to make sure there's no change. The Phoscon screenshot is missing the "-A" from the model identifier.

You could try a couple of options. Deconz may misss a sensor every now and then:
- If not already done, delete the device in Phoscon and join it again.
- Start sensor search again and hold the sensor in your hands, so it gets warmer. It may trigger the device to send temperature data
- Start sensor search again in Phoscon and the read the basic and/or temperature attributes in deconz GUI

tosion, 2020-03-29:
OK - thanks again!

Basically I have done those steps you suggested with no luck. Temperature attribute shows also 0 in Deconz GUI. Right now there in not too much I can do anymore? The sensor is located in my summer cabin and unfortunately because this corona we are not allowed to travel there from my home until it gets better ( I just yesterday came back from the cabin but I am not allowed to go back again ....). Remotely I can use Phoscon and Deconz GUI and delete it but rejoin is not possible remotely. 

So the sensor is perhaps broken?

SwoopX, 2020-03-29:
> So the sensor is perhaps broken?

Hard to say without physical access. Those sensors are sleeping most of the time and need to be woken up to get any data. While having access, you can force the device to wake up in some ways (e.g. clicking some physical buttons, warm it up, etc.). So, if you don't get any temperature readings via deconz GUI, the device is either sleeping or maybe not correctly paired. I that sense, I'm also not too sure any more whether information presented in your screenshots is actually correct. I'd have expected to see at least some fields greyed out...

tosion, 2020-03-30:
[image]

Excellent - You are a Hero!

After reading again your writing I understood that I can do the cluster check same time when adding a new sensor in Phoscon side. That did the trick! (My bad I did not read your text carefully first time).

Just to make sure the sensor worked correctly as motion sensor at very beginning ...

So now I have a working sensor  - both motion and temperature. Thanks!

SwoopX, 2020-03-30:
> Excellent - You are a Hero!

Finally someone recognizes it 😛 Just kidding. Glad you made it.

> My bad I did not read your text carefully first time

Happens to me way too often as well.

Would you mind closing the issue then? Hf and take care!

closed 2020-03-30 by tosion