dresden-elektronik/deconz-rest-plugin #2629: Smartthing siren not connecting

paulmann1590, 2020-03-29:
Hi all, 
is there any chance to get this Sirene running with Deconz? I bought it, but it always losing connection after 1 minute and no chance to connect it. 
https://www.smartthings.com/de/start/devices/smartthings-sirene

Thanks and best regards!

SwoopX, 2020-03-29:
Pretty sure the device is not yet supported. Please issue a device support request as described in the wiki, thanks.

paulmann1590, 2020-03-30:
Hi :) Thanks for your reply. I realized, that there was a request already done here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2051 But unfortunately it is closed. How can I help to get the device running :) ?

SwoopX, 2020-03-30:
Ah, ok. The disconnect smells like either device did not successfully pair or did not get properly enrolled. There's definitively some code amendments required to have it supported but I'm not too sure if that will solve the disconnects.

Would you feel comfortable with compiling the rest plugin yourself and do some tests?

paulmann1590, 2020-03-31:
Thank you very much for the quick reply and your try to help :) 

Comfortable, no ;). But I am willing to help, and if there is an how-to available I will try it - and also learn :). I have a windows computer available and the raspberry where the Hassio is running on. 
Is this enough, or do I need something else?

SwoopX, 2020-03-31:
Did a little investigation on the device. It looks like other projects also face toe issue with the disconnect, which is unresolved yet. From that perspective, we'd require a network sniff to further analyze communication.

Eventually, a comparable mechanism is used as with Legrand devices, but hard to tell without that information.

paulmann1590, 2020-03-31:
That's a pity. At least in Germany the device is currently available at a very cheap price and it would be a good alternative to other sirens available. 
I did some research regarding the sniffing, and it seems like I would need another device / stick to do it?

SwoopX, 2020-03-31:
Wanna join the discord channel to discuss further? Maybe you can send it over or I buy one and send it to you afterwards...

SwoopX, 2020-03-31:
Another idea since we have a comparable discussion for another device:

When you try to pair and the device shows up in deconz GUI, have you tried to read the basic cluster attributes?
If that doesn't work, pair the device until shows up in deconz GUI, push the reset button and read the basic cluster attributes.

Maybe that helps to hold the device in the network.

paulmann1590, 2020-03-31:
I just tried again to add it. When connecting it via deconz GUI it is added for some seconds. It shows also up with a green connection line in the Windows App. 
The time is enough to read the basic cluster.
[image]

But after some seconds the connection is lost, and I need to reset it and connect it again.

ebaauw, 2020-03-31:
The pairing process in the REST API plugin could be improved (to say it nicely).  It's an unholy sequence of requests (hopefully) leading to a response that triggers the next request.  There's no real orchestration of the sequence of requests, guarding it with timeouts and retrying.

I've written down somewhere in another issue what all the pairing steps (that I've identified) entail, but basically the following happens:
- The network is opened, i.e. the coordinator sends a periodic broadcast telling routers to allow new devices to join the network.
- The device joins the network (receives the network key) and issues a device announcement broadcast.
- The deCONZ core creates a node and reads the _Node Descriptor_;
- The deCONZ core reads the _Basic Descriptors_, and populates the right drop-down circle with the end-points and clusters;
- The REST API plugin creates the `/lights` resources, based on info from the descriptors.  When the resource is created, the label on the node changes from the NWK address to the resource name;
- The REST API plugin reads the _Basic_ cluster attributes;
- The REST API plugin creates the `/sensors` resources, based on info from the descriptors and the _Basic_ cluster.  When the resource is created, the label on the node changes from the NWK address to the resource name;
- The REST API plugin configures the device, using multiple requests, creating a `/groups` resource (for ZB3 devices), binding the client clusters to that group, binding server clusters to the coordinator, setting up attribute reporting, enrolling the IAS Zone, etc.
- Where appropriate, Phoscon creates rules for the device.

If one of the responses is missing, the pairing process stalls.  It can be restarted by manually triggering the response (reading the _Simple Descriptors_, reading the _Basic_ cluster attributes, etc.).  However, the device must be awake, have its radio powered, and still be in the network, to receive the manual request.  This might be challenging, especially for (battery-powered) end devices (grey nodes), or for devices that leave the network if not configured in time.

paulmann1590, 2020-04-01:
Thank you very much for the details. Unfortunately I am myself no developer, so I feel I can't support much here. But of course if I can provide information, I will do it :)
 @SwoopX I can join this channel later, need to download the app first :)

SwoopX, 2020-04-16:
Solution is to be found here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2051#issuecomment-614925239.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix