dresden-elektronik/deconz-rest-plugin #2630: Centralite Pearl Thermostat setup

stevewull, 2020-03-29:
I use deCONZ running on Windows with a Conbee II usb stick. I also use the deCONZ integration with Home Assistant. Works great. I mostly use Ikea and Aqara lights, sensors, and switches. deCONZ support for the Centralite Pearl Thermostat was added in the Beetlejuice release. (Issue #1601). 

(1) I installed this thermostat. It connected to my deCONZ network. But, the device has no name and is not visible in Phoscon. How do I add a name? Anything else that needs to be done in deCONZ? Will it appear in Phoscon?

[image]

(2) Further, the thermostat does not appear as a device in Home Assistant. What must I do in Home Assistant to get the thermostat and its sensors configured?

SwoopX, 2020-03-31:
Does the device show up in the API? Check out https://dresden-elektronik.github.io/deconz-rest-doc/ for additional information.

stevewull, 2020-03-31:
No. It does not appear in the API. I got the full state of the gateway and searched for the thermostat. It is not there.

As an aside, looks like I am using version 1.16.0 of the api and version 2.05.75 of deCONZ.

SwoopX, 2020-04-01:
Try reading the basic cluster while sensor search is active.

ebaauw, 2020-04-01:
> version 1.16.0 of the api 

That's there, because Hue clients expect it, but it isn't maintained.  There have been many changes to the API, without changing the API version.

stevewull, 2020-04-01:
If I understand correctly, I should use the Phoscon App to search for new sensors from an 'Other' vendor. While the search is going, I should use the deCONZ app to read the basic cluster, which I do by going to the cluster info and clicking 'read' in the attributes section. This does add a lot of information to the attributes:

[image]

The search for the sensor in Phoscon ends after 3 minutes displaying the message, "Failed to connect." The Thermostat still does not appear in the API.

SwoopX, 2020-04-01:
Thanks. This is a different manufacturer and modelID than previously. We've seen that on Centralite devices like twice. I double check and can add it if required.

stevewull, 2020-04-02:
I would certainly appreciate if you added it.

As for overall interest, hard to say. Looking the current list of supported thermostats, the Centralite is the only one available in the US. It would be good for the deCONZ platform to support a few Zigbee thermostats for North America. The Centralite is less expensive than the wifi models garnering so much hype and press. It does not require a connection to the cloud. It avoids the need to consider adding a z-wave network to the zigbee and wifi infrastructure. I could build my own with an ESP32 and some relays, but most people do not want to do that.

SwoopX, 2020-04-02:
Unfortunately, it's not that easy. We can integrate almost every device upon request if people are willing to provide all information required and do the required testing. Please note that we try to do that "remotely", so we don't have the device in our hands mostly. Integration can get rather technical and my also require to sniff the zigbee traffic to gain a better unterstanding, what's going on in the air. That especially applies if any manufacturer specific clusters or attributes come into play, requiring some reverse engineering or, ideally, analyze the communication with the vendor gateway. Apart from that, it is not exceptional that devices do not comply with the standards, adding some additional spice.

Pro grade devices quite often offer a detailed technical manual, which is always quite helpful, if publicly available. However, for most devices, that's not the case unfortunately.

So bottum line is that it's not always as easy as people think to get the devices working.

stevewull, 2020-04-02:
Yes, I've been working in the software and system integration industry for a long time. I empathize. This isn't a product that has much demand. There are other more important things to do.

I was hoping that since you already integrated the Centralite Pearl, this would require only slight modifications as opposed to a full new integration.

If you give this attention over the next two weeks, I'm happy to be a tester and supply whatever information I can. Otherwise, I am going to return it and move to a different thermostat solution.

Still very happy with deCONZ overall for my Ikea and Aqara devices.

SwoopX, 2020-04-03:
I guess we're about to misunderstand each other, sorry for that. I'll have the device covered within the next days, no worries.

Seems like I got "It would be good for the deCONZ platform to support a few Zigbee thermostats for North America." wrongly. Thought you were taking about others which nobody knows about here ;)

stevewull, 2020-04-03:
Ah. Got it.

Thank You.

stevewull, 2020-04-03:
@SwoopX   if I read the full state of the api and perform a search for a portion of the uniqueid ("cb"), I do not find any devices. It is not appearing in the API.

SwoopX, 2020-04-04:
Ok, thanks. Then we need to have a look at the code. Would compiling the plugin be an issue for you and are you using docker?

stevewull, 2020-04-04:
All of my linux (Ubuntu) boxes are hyper-v VMs that, unfortunately, cannot access the USB port of the host (without some 3rd party USB to IP solution). Therefore, I am currently using the Windows 10 version of deCONZ with my Conbee II.

Other options for test purposes would be for me to install everything on a Raspberry Pi 3 or install in a Docker container on a Synology DS718+ NAS. I've been tempted to move from the Windows box to the Synology anyway, if deCONZ is compatible. 

I'm comfortable learning the steps to compile.

SwoopX, 2020-04-04:
I'd say the Pi should be the preferred solution for the moment.

When you've something ready, make sure to add the deconz **dev** repository as described here: https://phoscon.de/de/conbee2/install . The instructions how to compile are to be found in the readme of this repo. However, after having cloned deconz (so step 1), download and overwrite de_web_plugin.cpp from here https://github.com/SwoopX/st_playground . It includes some changes which should do the trick. Then continue with the other steps as described.

Good luck!

stevewull, 2020-04-04:
Ok. Will try over the next few days and let you know. 

While we're at it, can we also test the Sengled E13-N11WA PAR38 bulb (similar to issue#2155). The light bulb works fine, but I am not seeing the lux and motion sensors.

SwoopX, 2020-04-04:
Sure, I'll have a look as well.

stevewull, 2020-04-05:
I installed the beta version of deconz on the pi.

Where are the instructions for compiling and cloning?

SwoopX, 2020-04-05:
https://github.com/dresden-elektronik/deconz-rest-plugin

Scroll down, after the source code.

stevewull, 2020-04-06:
Got it. I followed all of the steps to compile and install dev with your web plugin, although I'm not sure how to verify that it is using yours. Deconz is running. It found all 26 of my nodes. 

I ran the phoscon app in the web browser
a. I searched for lights. It updated all of my Ikea bulbs. It updated the two Sengled Par38 bulbs
b. I'm not sure how to add the existing switches and sensors to the phoscon app without repairing
c. More important for our purposes, the Centralite appears in the deconz interface as below. The option to see the cluster info is not there.

[image]

SwoopX, 2020-04-06:
Ok, so let's try it out. Please do the following:

1. Stop deconz from commandline "sudo systemctl stop deconz-gui"
2. Start deconz again with "deconz --dbg-info=2 --dbg-error=1 > thermostat"
3. In Phoscon, search for new lights
4. Physically reset the thermostat

Then, I need to have a look at the thermostat file created in 2.

From the latest screenshot, the device has not properly joined, not too sure why.

stevewull, 2020-04-06:
It gives me an unrecoginzed option '--dbg-info=2' error when I type: "sudo systemctl start deconz --dbg-info=2 --dbg-error=1 > thermostat"

SwoopX, 2020-04-06:
It was on purpose to omit "sudo systemctl start" in #2 because of that. We don't want to start the system service for this, but deconz directly ;)

stevewull, 2020-04-06:
but when I leave it off, it gives me the message deconz command not found.

SwoopX, 2020-04-06:
I guess, it could be case sensitive and I've put an alias on it.

Try "deCONZ" or search for it with "which deCONZ".

stevewull, 2020-04-06:
Here is the message I receive after running the command

[image]

I'll attach thermostat file to next post.

SwoopX, 2020-04-06:
Thanks. Took a look at the file and apparently, you've forgotten to exchange de_web_plugin.cpp with the version from my repository. This is really important as it contains more debug output and has some code changes for your device. Can you please give it another go?

stevewull, 2020-04-08:
I thought I had exchanged them.  I tried again. This time I get an error during compile:
de_web_plugin_cpp:7238:71: errr: stray '\342' in program
followed by some other strays \200, \255, \200

SwoopX, 2020-04-08:
Sorry, that one's on me. Copied some values in the without commenting them, fixed now. Can you please download and try again?

stevewull, 2020-04-08:
Now it gives me different errors: 
681:35: 'basicClusterEndpoint' was not declared in this scope  apsReq.setDstEndpoint(basicClusterEndpoint);

721.9: jump to case label [-fpermissive]

and some notes and warnings about two unused variables

SwoopX, 2020-04-08:
Compiled it myself now and got rid of the errors. The warnings can be ignored. Please download and try again.

stevewull, 2020-04-08:
Is this file helpful 
[termostat.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4453026/termostat.zip)

SwoopX, 2020-04-08:
Indeed. Check the REST API, you have an added thermostat sensor.

stevewull, 2020-04-08:
Yep. I see it in the API!

Now I need to set up a test instance of Home Assistant

SwoopX, 2020-04-08:
Cool. I quickly prepare a commit so it's added to my open pull request and hopefully find its way into the next release. Wanna also try the changes for your bulps?

stevewull, 2020-04-08:
Yes. I'll switch to that ticket  (#2667 )

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

cbrightly, 2020-09-06:
I recently purchased this same unit and it appears, again, quite similar, however I am experiencing identical issues as are described in the OP.
[image]
[image]

Please advise!

SwoopX, 2020-09-06:
@cbrightly Please raise a seperate issue for your current situation and provide all the requested information. This should give us a clearer picture on what might be going on (or not) on your end. Thanks!

cbrightly, 2020-09-06:
Done, thank you.