dresden-elektronik/deconz-rest-plugin #2631: [New Device Request] Blitzwolf BW-IS3 PIR Motion Sensor

baldus88, 2020-03-30:
I am running ConBee II, Deconz Addon in HASSIO and would request if possible to extend this device:
https://www.blitzwolf.com/Rechargeable-ZigBee-PIR-Motion-Sensor-p-442.html

I have searched and not found any open issue for this and not listed to supported devices.
I have launched the first way the deconz GUI via VNC but I don't find this device in deconz map, what is the right procedure for the view?

Could you please help me to attach the cluster and nod info requested by the specific page for the new device request?

Regards

SwoopX, 2020-03-30:
Have you tried pairing it via Phoscon sensor search? It should result in the device showing up in deconz GUI...

baldus88, 2020-03-30:
> Have you tried pairing it via Phoscon sensor search? It should result in the device showing up in deconz GUI...

Sorry "phoscon sensor search" is a function in deconz gui? 

This is my gui:
[image]

SwoopX, 2020-03-30:
Phoscon is the software on top of deconz. More precise, it is just another client for deconz' REST API. Usually, it is reachable on port 80 of the system you installed it.

So try http://YOURIPADDRESS and check it out. Btw, you should reset the sensor directly after you activated the sensor search.

SwoopX, 2020-03-30:
> Sorry but you don't mean the phoscon panel webapp to add the new devices right?
This is active on 40850.

That's indeed what I meant. Usually, it runs on port 80, but I don't know your setup.

> Don't find nothing.

Have you reset the sensor while the search was active? If yes, the device should have shown up in deconz GUI (very first screenshot you posted).

baldus88, 2020-03-31:
Hi.
When try to reset the device the sensor finally appears in map:

[image]

But when try to open the cluster info is empty, is normal or necessary to active some flags in gui?

Thanks
Regards

SwoopX, 2020-03-31:
Ok, that's a first step. Now, as to my experience, a restart of deconz should enable you to read the basic cluster. You should also try the re-read the simple descriptors (the right of the two dots from the device in your screenshot). If that doesn't help, repair again (eventually requires multiple times). Also make sure the sensor is close to your Conbee.

Especially cheap devices from China tend to be tricky here.

baldus88, 2020-04-01:
Hi.
I have reset the deconz GUI and now I have read different values of this device for the node, is normal? The NWK is 0xD927 instead of 0xA9CF. 
The IEEE of node info is the same: EC1BBDFFFE303275

This is the node info:
[image]

Basis - Cluster Info:
[image]

Cluster Info - Identify:
[image]

Cluster Info - OTAU Cluster:
[image]

Is ok to analyze?

Thanks!

ebaauw, 2020-04-01:
> I have reset the deconz GUI and now I have read different values of this device for the node, is normal?

It looks like the values were never read in the first attempt.  In particular, the _Basic Descriptor_ containing the device type and clusters.

The cluster list still seems incomplete, however.  There's no cluster through which the sensor can communicate that it detects motion.  I would expect a client (grey) _On/Off_ cluster, based on the device type.  Try re-reading the _Basic Descriptor(s)_ (from the left dropdown circle in the node), while waving in front of the sensor, to make sure it's awake, and see if additional clusters appear.  Otherwise you might need to sniff the ZigBee traffic to figure out how the sensor communicates.

> The NWK is 0xD927 instead of 0xA9CF.

That's the short address assigned to the device on joining the network.  It changes when you reset and re-pair the device.

SwoopX, 2020-04-01:
Additionally to what @ebaauw mentioned, the **device** should be physically reset; don't do it via deconz GUI. Keep the sensor awake by waving your hands in front of it while pairing. If you don't get a ready message from Phoscon while sensor search is active and the device has been reset, try also to re-read the simple descriptors continuing to force motion.

baldus88, 2020-04-02:
Hi.
I have try to restart deconz addon via hassio and then the nodes reconnect to mesh network:
[image]

But what is the right procedure to "try re-reading the Basic Descriptor(s)"??

I have to click on Node On/Off Switch -> Click on Cluseter 000 Basic (29) 
and Cluster Info -> Basis Cluster -> Attributes
and clic Read but I have viewed the same attributes to the past attach.

Is necessary other tool to sniff the zigbee traffic for this device?

ebaauw, 2020-04-03:
> But what is the right procedure to "try re-reading the Basic Descriptor(s)"??

My bad: the _Simple descriptor(s)_.  You read them from the left drop-down circle on the node.  Make sure to wake the device.

baldus88, 2020-04-04:
HI.
When try to click on "simple descriptor", now show a flag to enable/disable and not change nothing near the word "Read", the number 000 remain the same:

[image]

Is normal?
When clicking on "node descriptor" the counter change from 001 to 000.

[image]

Thanks!

ebaauw, 2020-04-04:
Did you wake the device?

baldus88, 2020-04-07:
Hi.
The device is online and wave my hands but don't show any additional descriptors.

[image]

I have tried to click on read simple descriptor, reboot the conbee ii, reread the attributes..

Do you have another workaround for this? :(

ebaauw, 2020-04-07:
Afraid not.

baldus88, 2020-04-08:
> Afraid not.

"Otherwise you might need to sniff the ZigBee traffic to figure out how the sensor communicates."
You have inserted this test, what is the procedure for sniff the zigbee traffic?

ebaauw, 2020-04-08:
I use ZShark to capture the traffic and WireShark to display/analyse it.  See the pinned issue for ZShark; there’s plenty of YouTube tutorials for WireShark.  You need a separate ConBee or RaspBee for ZShark; the capture firmware is not yet available for ConBee II and RaspBee II.  There’s also other capture solutions, using different Zigbee hardware, but I have no experience with those.

baldus88, 2020-04-08:
If you want or other users support me via remote desktop or similar I am available to investigate this and support the deconz team to development the devices or you have another way to support me.

Regards!

ebaauw, 2020-04-09:
See #2675: the Alecto smoke detector looks like a spitting image of this motion sensor from a Zigbee perspective.  No chance in hell, we'll be able to understand the smoke detector without access to their native bridge.  I fear the same holds for the motion sensor.

Alecto do offer a motion sensor: https://www.alecto.nl/smart-motion10-smart-zigbee-bewegingssensor-wit, but it appears to be a different model.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

baldus88, 2020-06-06:
> It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

how can I reopen the topic?

djgreko, 2020-07-27:
I purchased this sensor because I thought it was recognized by having another blitzwolf device which is seen as HEIMAN. But unfortunately it doesn't work. it is very convenient because it has its own internal USB rechargeable battery unlike xiaomi sensors that need a button battery.

wovka88, 2020-10-12:
I have also a BW-IS3 but is not working. Which information are necessary to integrate the device? How can i help?

Mimiix, 2020-10-12:
@SwoopX You re-opened but did not provide a why.

SwoopX, 2020-10-13:
Cuz it was a device support request which Smanar maybe could pick up.

Mimiix, 2020-10-14:
@Smanar

ASNNetworks, 2020-10-20:
So why is this closed? The device hasn’t beed added right? Or am I missing something?

Mimiix, 2020-10-21:
@ASNNetworks Nope it wasn't afaik. It was closed due to inactivity.

wovka88, 2020-10-21:
Can i help? I have the device at home. Which Information are necessary?

Mimiix, 2020-11-02:
@wovka88 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2631#issuecomment-610856946 tells what is needed for it.

Nailik, 2020-12-10:
Zigbee2Mqtt supports this device.
https://www.zigbee2mqtt.io/devices/BW-IS3.html

Sadly Zigbee2Mqtt still has some issues with the deconz adapter, that's why i don't use it. But maybe you can use the information there to support it in deconz.

There is the pull request when they added it:
https://github.com/Koenkk/zigbee2mqtt/pull/2821

Haasie, 2020-12-17:
Hello, could you please Ad support for this device?
Zwitched front cc2531/Z2m to conbee/deconz but Deens deconz doesn't support the IS3 het.

Best regards.

Haasie, 2021-01-01:
any changes ?

Mimiix, 2021-01-01:
@haasie provided info is still not there. @Smanar @SwoopX can you use the info from z2m?

Haasie, 2021-01-01:
Maby i can help?

SwoopX, 2021-01-02:
I don't touch this.

However:
[image]

Haasie, 2021-01-24:
Still nothing?

Mimiix, 2021-01-25:
@Haasie We still don't have the info needed. Sniffed would be best.

Haasie, 2021-01-25:
I would like to collectie te info, if i know how.

closed 2021-04-04 by github-actions[bot]

djgreko, 2021-05-04:
I have read about the latest deconz update with many additions but unfortunately this sensor has not been added.

Haasie, 2021-07-31:
still nothing

hubirolf, 2021-09-20:
Habe hier die selben Probleme. Der BW- IS3 wird nicht eingebunden.