dresden-elektronik/deconz-rest-plugin #2635: Failed to start deconz-gui.service on raspberry

bruesche, 2020-03-31:
Hi there,

I try to start the deconz-gui.service on raspiannbuster.  The little brother (deconz.service) is working fine. I stopped the deconz.service and started the deconz-gui.service. Status of the service is:
pi@raspberrypi:/lib/systemd/system $ sudo service deconz-gui status
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-03-31 16:20:26 CEST; 12s ago
  Process: 3046 ExecStart=/usr/bin/deCONZ --http-port=80 (code=exited, status=1/FAILURE)
 Main PID: 3046 (code=exited, status=1/FAILURE)

I looked for the service file and saw that the packages lighted and VNC are needed. VNC is working well. I can connect via a virtual desktop and tried to start the gui. Nothing happens. I looked for lightdm. There I had an issue to. Missing package accountservice. Now this is corrected too.

But I can't still start the deconz-gui.service. Same failure you can see above. 

Can anybody give my advice to solve the problem?

Regards 
Thorsten

ebaauw, 2020-03-31:
> Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled

Did you try to enable the service?

bruesche, 2020-03-31:
Now trying to enable:
pi@raspberrypi:/lib/systemd/system $ sudo service deconz-gui status
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-03-31 17:47:20 CEST; 6s ago
  Process: 6668 ExecStart=/usr/bin/deCONZ --http-port=80 (code=exited, status=1/FAILURE)
 Main PID: 6668 (code=exited, status=1/FAILURE)
pi@raspberrypi:/lib/systemd/system $

ebaauw, 2020-03-31:
Do you see any messages in the syslog that might give a clue?  `sudo journalctl -u deconz-gui`

bruesche, 2020-03-31:
Mär 31 16:13:23 raspberrypi deCONZ[10051]: Could not connect to any X display.
Mär 31 16:13:23 raspberrypi systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
Mär 31 16:13:23 raspberrypi systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Mär 31 16:13:53 raspberrypi systemd[1]: deconz-gui.service: Service RestartSec=30s expired, scheduling restart.
Mär 31 16:13:53 raspberrypi systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 5.
Mär 31 16:13:53 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Mär 31 16:13:53 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Mär 31 16:13:53 raspberrypi deCONZ[11988]: qt.qpa.screen: QXcbConnection: Could not connect to display :0
Mär 31 16:13:53 raspberrypi deCONZ[11988]: Could not connect to any X display.
Mär 31 16:13:53 raspberrypi systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE
Mär 31 16:13:53 raspberrypi systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
Mär 31 16:14:23 raspberrypi systemd[1]: deconz-gui.service: Service RestartSec=30s expired, scheduling restart.
Mär 31 16:14:23 raspberrypi systemd[1]: deconz-gui.service: Scheduled restart job, restart counter is at 6.
Mär 31 16:14:23 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Mär 31 16:14:23 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Mär 31 16:14:23 raspberrypi deCONZ[13834]: qt.qpa.screen: QXcbConnection: Could not connect to display :0
Mär 31 16:14:23 raspberrypi deCONZ[13834]: Could not connect to any X display.
Mär 31 16:14:23 raspberrypi systemd[1]: deconz-gui.service: Main process exited, code=exited, status=1/FAILURE

bruesche, 2020-03-31:
Don‘t Know what to do. Can you help me?

ebaauw, 2020-03-31:
> Mär 31 16:13:23 raspberrypi deCONZ[10051]: Could not connect to any X display.

That would crash the GUI alright.  Are you running the graphical environment on the Raspberry Pi?  You don't need a connected display (use VNC), but the graphical environment needs to be up and running.  You did not modify `deconz-gui.service`, did you?

bruesche, 2020-04-01:
I use vncserver with a virtual desktop. Works fine. I did't modify the deconz-gui.service. Tried with another http Port but this was no solution.

bruesche, 2020-04-01:
I want to start the desktop gui because I found an article to connect the eurotronic spirit zigbee thermostat with deconz (and Homebridge). Found this tutorial https://my-darling-is-a-linux-user.org/?p=827 and therefor I need the gui.

ebaauw, 2020-04-01:
> I use vncserver with a virtual desktop. Works fine. I did't modify the deconz-gui.service.

There seems to be something wrong with your system configuration.  Sorry, this is beyond me.

You might try and start deCONZ manually, from a terminal window in VNC.

bruesche, 2020-04-02:
It woks! I started the virtual desktop and in the session the terminal window. Stop deconz.service and start deconz-gui.service without problems. Could start the deconz app and the tutorial for eurotronic thermostat works well. 
Thanks for your help!

closed 2020-04-02 by bruesche

hesspoint, 2021-05-05:
I tried everything. Cannot open the Deconz app via VNC. I click on Programming>Deconz and just nothing happens. No error message. Actually when I connect the Pi to a screen and keyboard the app opens. Just not on VNC.

Tried starting Deconz and Deconz-GUI inside VNC via the terminal. Does not work. Tried stopping Deconz and starting Deconz-GUI like @bruesche did. Still does not work

Always get that error that @bruesche  reported also earlier. What am I doing wrong?

sudo service deconz-gui status
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2021-05-05 21:49:29 CEST; 14s ago
  Process: 23672 ExecStart=/usr/bin/deCONZ --http-port=80 (code=exited, status=1/FAILURE)
 Main PID: 23672 (code=exited, status=1/FAILURE)

johnthackers, 2021-11-19:
> Hi there,
> 
> 
> 
> I try to start the deconz-gui.service on raspiannbuster.  The little brother (deconz.service) is working fine. I stopped the deconz.service and started the deconz-gui.service. Status of the service is:
> 
> pi@raspberrypi:/lib/systemd/system $ sudo service deconz-gui status
> 
> ● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
> 
>    Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
> 
>    Active: activating (auto-restart) (Result: exit-code) since Tue 2020-03-31 16:20:26 CEST; 12s ago
> 
>   Process: 3046 ExecStart=/usr/bin/deCONZ --http-port=80 (code=exited, status=1/FAILURE)
> 
>  Main PID: 3046 (code=exited, status=1/FAILURE)
> 
> 
> 
> I looked for the service file and saw that the packages lightdm and VNC are needed. VNC is working well. I can connect via a virtual desktop and tried to start the gui. Nothing happens. I looked for lightdm. There I had an issue too. Missing package accountservice. Now this is corrected too.
> 
> 
> 
> But I can't still start the deconz-gui.service. Same failure you can see above. 
> 
> 
> 
> Can anybody give my advice to solve the problem?
> 
> 
> 
> Regards 
> 
> Thorsten

Good morning my conbee 2 started crashing like attached video could this be something todo with GUI interface 

Thanks in advance 

https://user-images.githubusercontent.com/27022133/142573966-6a90a3c3-363f-4671-978b-cf58f825c653.MOV