dresden-elektronik/deconz-rest-plugin #2636: hue dimmer switch delays

jonskoglund, 2020-03-31:
Hello. I am using Deconz as an add-on with Home Assistant. To activate a scene i want to use the Hue dimmer switch (type RWL021) with firmware 6.1.1.28573. But I can't get it to work properly. There's no response a button is pressed or a loooooooong delay :-( I'm using the deconz_event in my automation. A code sample of one of the buttons of this dimmer:

[image]

And this is an overview of the Zigbee Mesh

[image]

The node that has the problem is 'Hue Dimmer Switch1' (001788010219D0FF). Another switch that I use is a tradfri remote. That one works flawless with deconz.
Deconz log
```
22:47:42:847 no button map for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021
22:47:42:847 ZCL attribute report 0x001788010219D0FF for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:47:45:478 device state timeout ignored in state 3
22:47:58:565 device state timeout ignored in state 3
22:47:59:842 device state timeout ignored in state 3
22:48:01:574 device state timeout ignored in state 3
22:48:03:064 device state timeout ignored in state 3
22:48:04:065 device state timeout ignored in state 3
22:48:05:064 device state timeout ignored in state 3
22:48:07:340 device state timeout ignored in state 3
22:48:09:116 dev /dev/ttyAMA0
22:48:10:371 device state timeout ignored in state 3
22:48:14:065 device state timeout ignored in state 3
22:48:15:565 device state timeout ignored in state 3
22:48:17:064 timeout on poll APS confirm
22:48:17:840 device state timeout ignored in state 3
22:48:19:573 device state timeout ignored in state 3
22:48:21:865 drop request 13 send time 356, cluster 0x0008, after 278 seconds
22:48:27:086 Current channel 25
22:48:30:682 ZCL attribute report 0x000D6FFFFE45392F for cluster: 0x0000, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:48:32:239 device state timeout ignored in state 3
```
I also have delay issues with my Hue Motion sensor (0017880102030142). Which gives similar lines in the deconz log. 
```
23:01:19:595 no button map for: SML001 ep: 0x02 cl: 0x0402 cmd: 0x0A pl[0]: 000
23:01:19:595 ZCL attribute report 0x0017880102030142 for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
23:01:26:341 device state timeout ignored in state 3
23:01:29:565 device state timeout ignored in state 3
23:01:30:073 Current channel 25
23:01:31:565 device state timeout ignored in state 3
23:01:35:342 device state timeout ignored in state 3
23:01:36:392 device state timeout ignored in state 3
```
Any help or advice on how to fix this?

**Deconz Add-on**
5.3.2
**Conbee II Firmware**
2.05.75 / 08/03/2020
264A0700
**Hassio** 
0.107.4

jonskoglund, 2020-04-02:
Did some research on github and found this:
_Originally posted by @esseki in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/354#issuecomment-443535893_
> * press : 1000
> * simple press release : 1002
> * double press : 1004
> * treble press : 1005
> * quadruple press : 1006
> 
> It means that part of the latency I was reporting in this issue was not due to the rule being executed in the rule engine but due to the rule being fired on the "simple press release" (1002) event rather than on the "press" event (1000).
> 
I was using the simple press release codes (1002, 2002, etc) for the dimmer switch. Changed all codes for every button to 'press'. There's a lot less latency now with this new press codes. I will report back if I experience any delays.

jonskoglund, 2020-04-04:
Reported similar issues with timeout/delays before [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2175). I will remove reported 'faulty' zigbee devices and see if that removes this message:  `device state timeout ignored in state 3`

jonskoglund, 2020-04-04:
About the hue dimmer....
```
13:25:06:757 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000
13:25:06:757 ZCL attribute report 0x0017880102030142 for cluster: 0x0400, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
13:25:07:247 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
13:25:07:757 Bind response success for 0x0017880102030142 ep: 0x02 cluster: 0x0001
```
See the third line. End-device might sleep. Is that normal behaviour?

jonskoglund, 2020-04-04:
Found this comment by @manup via [this issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2333#issuecomment-587587216):
Maybe this issue is related with that. 
> One thing that we working on a while now is to address the software issues which generate the most support requests, like ConBee not connected after restarts and "whole network lost" kind of issues. Hopefully the updates will be out soon and relax the issues.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix