dresden-elektronik/deconz-rest-plugin #2638: Is possible to add more group state in websocket response (bri, ct, ...) ?

AlexpFr, 2020-04-01:
Hello,
When a change of state is received by the websocket client for a group, is it possible to add other parameters such as "bri" or "ct"?
For example, currently in case of change of "bri", we only receive information from the lights individually.
Thank you in advance.

ebaauw, 2020-04-01:
No, a group state only contains `all_on` and `any_on`.

AlexpFr, 2020-04-01:
Yes but is it possible to improve the websocket server to make a more complete group state return ?

It works on Phoscon but I think it uses the individual state of the lights to deduce "bri" and "ct".

ebaauw, 2020-04-01:
The problem is not with the web socket server.  The problem is that there is no group state.  Phoscon probably echoes the last action that was set on the group, but that's not the (average) state of the lights in the group.  Trying changing the state of the individual lights in the group and notice that Phoscon doesn't update the group "state".

AlexpFr, 2020-04-01:
Yes that's correct, websoket could also return the last action on the group in the state parameter.
It would be very useful for the spots management because the piloting is mainly done with the groups and not individually.

ebaauw, 2020-04-01:
> websoket could also return the last action on the group

Yes.  We would need to setup Events for changes to the group action attributes, with the added perk that you could also us these in as rule conditions.  Technically possible, but quite some work.  I will have other priorities for a long time.

> in the state parameter.

No bloody way!  That would cause inconsistencies between the API and the web socket notification.

AlexpFr, 2020-04-01:
I just tested changing "bri" on one of the light of the group, Phoscon considers this change and assigns the new value "bri" on group slider without changing "bri" on other bulbs of this group.
This is the expected result. I will try to reproduce this on my home automation control software (jeedom) while waiting for an implementation to be made in the websocket server.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

ebaauw, 2020-06-05:
No it's not revolved.  Low priority, though.

closed 2020-07-04 by stale[bot]