dresden-elektronik/deconz-rest-plugin #2639: [Device Support Request] LivingWise LVS-ZB-SC7 Smart Scene Controller with 7 Programmable Switches

jeepotvin, 2020-04-01:
### Device info
Model: LVS-ZB-SC7
Brand: LivingWise
Type: Smart Scene Controller with 7 Programmable Switches

### Where to buy
Sold on [PrimeCables](https://www.primecables.ca/p-368984-lvs-zb-sc7-neutral-zigbee-smart-7-button-scene-controller-us-type-neutral#sku381087). Seem to be identical to the [Orvibo brand](https://www.primecables.ca/en/p-360932-cab-t10s7zw-1uo-all-orvibo-zigbee-smart-scene-controller-with-7-programmable-switches), also sold on PrimeCables but with a much higher price tag.

### Cluster info
**Basic**
[image]

**Scenes**
[image]
[image]
[image]
[image]

### Node info
[image]

### Debug output
I tried the deCONZ app on a Windows machine and can't seem to get the logs out. Is there a way to get it out or I must use a Linux system for it?

### Device Image
[image]

jeepotvin, 2020-04-11:
Since I own a ConBee II device and it is [not yet supported](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/405#issuecomment-612349194) by ZShark, can I help by providing more information or log files?

Bitmaster20, 2020-04-18:
I have a couple of these and would like it supported also.  If you need any logs or testing about the device I can help also.

jeepotvin, 2020-05-04:
> I have a couple of these and would like it supported also. If you need any logs or testing about the device I can help also.

In the meantime, how do you use them? You must have some Living Wise or Orvibo bulbs compatible with the proprietary hub?

Garfield9988, 2020-06-02:
Hi All,
I have install 10 x ''LVS-ZB-SC7'' in SmartThings with success. By using device handler from Brad Sutton ''Orvibo T10S7ZW 7 Button Scene Controller'' and ''smart lighting'' in SmartThings to associated keypad fonction. The only problem is the LED indicator not showing statue. Any suggestion ?

Link: device handler: https://github.com/bradlee-s/det-SmartThings/blob/master/Orvibo%207%20Button%20Scene%20Controller.groovy

Best Regards
Mart

jeepotvin, 2020-06-02:
> Hi All,
> I have install 10 x ''LVS-ZB-SC7'' in SmartThings with success. By using device handler from Brad Sutton ''Orvibo T10S7ZW 7 Button Scene Controller'' and ''smart lighting'' in SmartThings to [...]

Hi Mart, 
Great to know these switches work with Smart Things. Unfortunately, I don't currently own one of those hub so I can't test nor help you out. I do hope though that it can help get it supported by deconz.
Regards.

jeepotvin, 2020-07-03:
Hey! Still interested to help get support for this device. 🙏

sktaylortrash, 2021-02-19:
Re-opening this request. As I don't see it in supported devices and would also like to use this unit. If information is missing from the request please advise.

Mimiix, 2021-02-19:
@SwoopX @Smanar What do we need to make it happen?

sktaylortrash, 2021-03-22:
Bumping to keep open would still like support for this device

jeepotvin, 2021-04-25:
Still would like to help add compatibility of this device. What are the tools I should install to help? Any documents at this point would be helpful.

sktaylortrash, 2021-05-23:
Yet another bump - what do we need to provide to get help with this device?

jeepotvin, 2021-05-25:
It's unclear how the community can help. Last time I checked, the ZShark tool wasn't supported for ConBee II.

Since then, I moved all my zigbee network to ZHA. Guess I should ask elsewhere how to contribute now.

jeepotvin, 2021-06-16:
Well I won't be keeping this issue active every month so I guess this is one last time. Again, I'm interesting in helping, it's just a little bit overwhelming and unclear what the community can do at this point.

sktaylortrash, 2021-07-21:
come on guys commenting a status more than once every five months shouldn't; be that difficult

jeepotvin, 2021-07-21:
> come on guys commenting a status more than once every five months shouldn't; be that difficult

For what purpose?

sktaylortrash, 2021-07-21:
Well a developer commented 5 months ago and then never again

Mimiix, 2021-07-21:
I'm not a developer :)

Mimiix, 2021-07-21:
Happy to re-open, but a dev needs to take care of it.

jeepotvin, 2021-09-30:
☝️

jeepotvin, 2021-11-15:
🧸

jeepotvin, 2021-12-11:
👁

closed 2022-01-10 by github-actions[bot]