dresden-elektronik/deconz-rest-plugin #2640: Gledopto GL-B-008ZS bulb - deconz bug - brightness below 13% = off

capstan1, 2020-04-01:
Hello,

I've just got these gledopto colored bulbs: GL-B-008ZS. Connecting them to deconz was easy, but a weird issue appeared connected to brightness:

- brightness <13% will turn off the bulb
- when turning the light on to for example 12% it flashes shortly, then switches off
- when dimming down to a value <13% it starts dimming and then turns off
- this leads to weird behavior when turning a light on to higher brightness when it was previously below 13%: it flashes briefly, turns off, then when the dimming reaches 13% it turns on again and dims until the brightness is reached

Here is what I got from the VNC viewer:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Let me know if you need anything else!

ebaauw, 2020-04-01:
Can you try in the GUI to issue _Move to Level_ commands and _Move to Level (with On/Off)_, with different levels (above and below the 13% threshold (~33) and with different transition time (at least 0 and 4, which is the default that the REST API plugin uses.

How do you control the lights?  Do you know what API commands are sent?

capstan1, 2020-04-01:
Thanks for your fast reply!

- Move to Level: All brightnesses (1-255) work with all transition times (0-5s), none turns it off
- Move to Level (with On/Off): Brightness from 2-255 work with transition time 0, brightness 0 and 1 turns the light off. Brightness 0-32 turns the light off when using transition time >0

So the problem seems to be in the transition time when using level with on/off.

I'm controlling the lights with Home Assistant wie deconz Addon normally. But the same problem exists when controlling the lights via Phoscon UI. I don't know about the API commands, where can I see this?

ebaauw, 2020-04-01:
> Move to Level (with On/Off) [..] Brightness 0-32 turns the light off when using transition time >0

That would be a bug in the bulb firmware.

> I don't know about the API commands, where can I see this?

Easiest in WireShark, sniffing the traffic between Phoscon/HA and deCONZ.  They're probably sending `{"on": true, "bri": #}` which the REST API plugin translates to a _Move to Level (with On/Off)_ and a default transition time of 4 (0.4 seconds).  With my PR refactoring of setting the light state (#2553), those commands will no longer be sent, and the firmware bug would be avoided.  Still need to do setting the group action after that, though.

capstan1, 2020-04-01:
Installed Wireshark but have no idea how to sniff the traffic 😄 Any directions?

So, at the end all I'd need to do is waiting for your PR to get merged and update?

ebaauw, 2020-04-01:
> Installed Wireshark but have no idea how to sniff the traffic 😄 Any directions?

Google "wireshark tutorial" - plenty of explanations on Youtube.  You want to be looking for HTTP traffic to the IP address of the machine running deCONZ.

> So, at the end all I'd need to do is waiting for your PR to get merged and update?

I hope so.

capstan1, 2020-04-06:
If I did everything correct it seems to send what you thought and transition time additionally.
[image]

ebaauw, 2020-04-06:
My PR got merged yesterday, so this should be worked around in the next version.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix