dresden-elektronik/deconz-rest-plugin #2641: Switch "Trust ZYCT-202" paired successfuly but not listed in the Photoscon

info-path, 2020-04-01:
Hello, my switch "Trust ZYCT-202" has been paired with Photoscon successfuly and can be used in a Home assistant. Unfortunately, it not listed at all in the Photoscon. I thought it was solved based on [this issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/752#issuecomment-460062095), but nothing was showing in the sensors/switches list. Sensors/switches from other manufacturers are listed there.

SwoopX, 2020-04-01:
The issue you referenced was for general support to allow for full functionality. However, it does not include anything related to Phoscon; it is closed source and the device will not show up there unless explicitly included by DE.

info-path, 2020-04-01:
Thanks for the information.

closed 2020-04-01 by info-path