dresden-elektronik/deconz-rest-plugin #2642: [Device Support Request] Shyne Smart Filament LED

hagenuck1, 2020-04-01:
Hi there, 
Would be great to have the Shyne Filament LEDs from click-Licht supported und Deconz. 
They’re working fine for me, but aren’t listed in supported devices till now.

[image]
[image]
[image]

Hope this is enough info. 
Turn on/ off, dimming and changing color temperature is working fine. 

I‘m using the Edison styled one.

SwoopX, 2020-04-01:
Well, if it works without any issues, there's probably nothing left to do for us here.

hagenuck1, 2020-04-02:
Thanks, then it should be listed as compatible at least for other users.

SwoopX, 2020-04-02:
Feel free to add them ;)

closed 2020-04-13 by hagenuck1