dresden-elektronik/deconz-rest-plugin #2643: Eurotronic Spirit ZigBee add External Sensor to API

KartoffelToby, 2020-04-02:
I have a RaspBee Setup running on 2.05.74 / 25.2.2020 Firmware 26340500
The setup for the Eurotronic Spirit ZigBee works like a charm discriped here: #1098

my question is, are there any plans to export the remote sensing flag to the rest api (v2)
and are there any solutions to connect a Xiaomi  Multisensor as an External Sensor to the Spirit?

I found that the build in Sensor is kind of useless.

I tried to ajust the temperature values with the api offset but it ends in "Err" after hours on the device. Maby because the temperature updates after 10 sec or so to the final offset val. and crashes in a wrong setting or overflow. This error ist kind of wierd. But without the offset ajustment the setup runs since days without problems.

SwoopX, 2020-04-02:
Have you checked #2399? It might not work as you expect it.

ebaauw, 2020-04-02:
> my question is, are there any plans to export the remote sensing flag to the rest api (v2)

No, as the setting doesn't seem to do anything.

> are there any solutions to connect a Xiaomi Multisensor as an External Sensor to the Spirit?

We haven't been able to find out how to tell the TRV to use an external sensor.

> I tried to ajust the temperature values with the api offset but it ends in "Err" after hours on the device.

Haven't seen that before.  Granted, I only used `config.offset` briefly while testing.  The `offset` should be stored to the _Local Temperature Calibration_ attribute (0x0010) and handled by the TRV itself.  I originally limited the offset (arbitrarily) to +/- 5 degrees, but they recently changed that.  Not sure if "extreme" values will be handled correctly.

KartoffelToby, 2020-04-16:
OK i have a solution with the WebSocket API and a NodeJS Script. Maybe it will help others:
```typescript
...
const subject$ = webSocket({
    url: `ws://${ip}:8088`,
    WebSocketCtor: WebSocket
    });
subject$.subscribe(
    (msg:any) => {
        if(type === 'sensors' && event === 'changed'){
            // GET DEVICES 
            // sensor (External Xiami Multi Sensor)
            // thermostat (Euronics Spirit ZigBee)
            ...
            if(thermostat.config.mode === 'auto') {
                const offset = Math.round(parseInt(sensor.state.temperature) - (parseInt(thermostat.state.temperature) - parseInt(thermostat.config.offset)));          
                this.deconz.setThermostat({
                    offset: offset
                });
            }
        }
   }
);
```

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue.

closed 2020-06-05 by Mimiix

mangoschorle, 2023-11-12:
This is a interesting thread. I made the unfortunate mistake of buying a Spirit on ebay and realized that the firmware 22190930 was buggy. Luckily with the help of [this thread](https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/issues/1) (@ebaauw and @KartoffelToby also contributed to this thread), I was able to manually flash a older version of the image.

Now to my problem:
I realized that the onboard temperature readings are not a good reference to base the control loop on. It just completely overshoots the setpoint. I will get a temperature sensor to see the actual roomtemperature in the upcoming days.
[image]

1. Has anything changed regarding the external measurements? Or is providing a offset still the only viable solution?
2. @KartoffelToby were you happy with your approach? And, where do you run that script? Simply on a RPI-alike system?
3. Are there good alternative supported thermostats by now? Which maybe even let you easily add external temperature readings, and maybe, or fancy stuff like room-size-dependent calibration routines?

I could theoretically use the manual mode of the spirit and write a separate controller running on dedicated hardware. I see many potentially very expensive screw-ups that could happen, so I really want to avoid this.