dresden-elektronik/deconz-rest-plugin #2644: Conbee changed name to Unkown 17 and I lost control of all devices temporarily

kses, 2020-04-02:
Hi!

The Conbee stick changed it name and shows up as a light with name unknown 17 and I can not control anything or get any sensor input. Conbee 1 running in a docker.

[image]
[image]

For a while I got this error every where:
error APSDE-DATA.confirm: 0xE1 on task

But know it started working after a few restarts... Happened yesterday evening as well.

ebaauw, 2020-04-02:
See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log: radio interference.

kses, 2020-04-02:
> See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log: radio interference.

Thank you! That is interesting... Do the channels are the same as for wlan? Will see if I can check the radio with my unifi AP

What about the unknown 17 lamp issue (Conbee USB drive)

ebaauw, 2020-04-02:
Interference is most likely caused by USB3.

ConBee should be listed as a Configuration Tool.  See #2534.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix