dresden-elektronik/deconz-rest-plugin #2645: Does xbee works on all deconz devices?

alekslyse, 2020-04-02:
I am trying to use a xbee device s2c  (XB24C) with the Zigbee TH Reg firmware - the reason is to be able to eleminate the xiaomi dropdout commands.

I set all the settings as stated in the FAQ, but its not possible to pair it - from the documentation the device should pair on reboot or config save, but nothing. 

Im not sure if its because im using the raspberry shield and not conbee 1 or 2 right now.

If anyone using the same xbee can you please export the profile from XCTU?

Getting this devices to work as routers really can help on the zigbee stability

ebaauw, 2020-04-03:
Did you see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2472?  Using those settings, the my XBees join the network, both RaspBee as well as ConBee II.  XCTU reports XB24C and firmware 4061.  The joining is a bit iffy - I found I can only join them while connected to XCTU.  After that, the rejoin automatically when power-cycled or reset.

Never created a profile before, but I hope this will help you:
[profile_4061.xpro.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4426152/profile_4061.xpro.zip)
I've had some issues that the XBee would leave the network when power-cycled, but with these settings it no longer does it.  Not sure what is the magic setting.  Also I had some issues that the XBee would always accept new devices onto the network, but with these settings (I think `JN` in particular) it doesn't.

In my experience, the XBees have a very stable ZigBee stack, acting as routers (repeaters) in the network.  One of them is even acting as parent to one of my Eurotronic Spirit TRVs.  At the moment, any devices attached to the XBee aren't exposed through the REST API.  Manuel provided me with the sources of an old XBEE plugin, see https://github.com/ebaauw/deconz-xbee-plugin, but I haven't found the time yet to look into that.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix