dresden-elektronik/deconz-rest-plugin #2646: [Device Support Request] NYCE NCZ-3014 Garage Door Tilt Sensor

mccracken, 2020-04-03:
NYCE NCZ-3014 Garage Door Tilt Sensor

[image]

[image]

mccracken, 2020-04-05:
[image]

mccracken, 2020-04-05:
> Can you please provide further screenshots as described in the wiki for requesting device support? Thanks!

Fixed now - thanks.

SwoopX, 2020-04-06:
The device should already be supported.

mccracken, 2020-04-08:
Doesn't seem to want to work. Any thing else I can try to test it out?

SwoopX, 2020-04-09:
What makes you believe it didn't work out. From the screenshot you posted earlier, I can see that pairing went alright and the simple descriptors have been read. Please don't tell me it's just because you cannot see it in Phoscon. You must check the REST API to be sure and verify sensor creation.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix