dresden-elektronik/deconz-rest-plugin #2647: Xiaomi mijia door/window sensor

Kristian8606, 2020-04-03:
The door sensor of Xiaomi mijia often does not respond and sometimes even disappears from deconz gui

Sjonnie2018, 2020-04-03:
Have you checked the device close(r) to your ZigBee receiver? Is the battery level of the sensor OK?

Kristian8606, 2020-04-03:
The battery is good 95%.  i didn't bring the sensor close to conbee because it is currently working and the problem arises when the door is open and i have to restart deconz then it does not respond until i remove it and add it again

Kristian8606, 2020-04-04:
In fact, the problem with the sensor manifests itself at all times.  Today for no reason he is not responding again

Mimiix, 2020-06-05:
@Kristian8606 is this still a issue?

Kristian8606, 2020-06-05:
Yes last night I restart my raspberry pi and the sensor stopped responding.  12 hours later, he began counting the opening and closing of the door again.

closed 2020-07-03 by stale[bot]