dresden-elektronik/deconz-rest-plugin #2648: device support: icasa remote ICZB-RM11S

jdeluyck, 2020-04-03:
Hi,

I recently acquired an iCasa remote - https://www.icasa.io/afstandsbediening?lang=en

Managed to pair it with deconz - it doesn't seem to show up in the phoscon, but I can receive events off of it. Shows up in the zigbee network. 

It seems to be made up of 4 "home automation level control switch" clusters.

jdeluyck, 2020-04-03:
I'm a bit puzzled, though, that the main buttons spam a bunch of events, and never the same ones. I need to properly take a dump of the output to see what it is actually sending through.

The specific group buttons react as I would have expected - x002 on short press, x003 on long press start and x001 on long press stop.

ebaauw, 2020-04-03:
The numbered _On_ and _Off_ buttons send a single command, to the group bound to the corresponding endpoint.  The main _On_ and _Off_ buttons send four commands: one to each group.  So these result in four button events per press.

Did you pair the remote under 2.05.75?  It's important that the REST API plugin set up the groups, or the main On/Off buttons don't send the commands from the right endpoints.  This causes the REST API to get confused, update `config.group` constantly, and miss out on some button events.  See #1978.

jdeluyck, 2020-04-03:
Yep, I just paired it, under 2.05.75.
Is there anything planned to get it to show up in Phoscon?

ebaauw, 2020-04-03:
You’ll have to ask dresden elektronik; Phoscon is not open source.

jdeluyck, 2020-04-04:
Did that ;)

Hypfer, 2020-04-19:
So whats the current status of this remote?
I did see this thread which states that the scene buttons aren't usable on their own. Is that information correct?
https://community.home-assistant.io/t/icasa-remote-with-deconz/158584

[image]

Do the other keys have unique keycodes now? Basically I'm looking for something to call some predefined scenes. I'd use custom code for it so I just need the events to work properly.

ebaauw, 2020-04-19:
Full REST API support with https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2538.

closed 2020-04-23 by jdeluyck

rweise, 2020-06-27:
I have a big problem with this remote. i can connect it and after this i found 4 Buttons in deconz but only the first one have a buttonevent. 
[image]
[image]

spiff42, 2021-01-14:
I don't know if this is the best place to ask for advice, my apologies if it is not (and please point me to a better place, I couldn't find any forum or similar).
I searched the Deconz supported device list in order to find a suitable handheld remote to use with deconz + OpenHAB. The one I found most suitable was the iCasa Remote. I first looked at Amazon, but it was not currently available, then tried all the sites listed at https://en.icasa.io/winkels, but none of them allowed shipping to Denmark. Then I googled and found a German webshop: https://www.ersatzteileshop.de/iczb-rm11s-zigbee-fernbedienung but unfortunately they were also unable to ship to Denmark.
I finally found this product on a Danish website: https://www.plusled.dk/zgr904-s-zigbee-fjernbetjening-touchlink.html. It looked exactly like the iCasa remote, so I thought it might be an OEM version (rebranded). I ordered it, and made an initial attempt to pair it with deconz (add new switch from Phoscon app web interface), but no success.
Then I managed to use touch-link pairing to get the remote to control an IKEA bulb.
And after finding the manual for the iCasa remote (https://www.easyfikssmarthome.nl/download/iczb-rm11s-remote-en.pdf) and reading about pairing with a Zigbee gateway which is not mentioned in the ZGR904 manual (https://www.plusled.dk/amfilerating/file/download/file_id/79/), and a lot of attempts with factory reset and pairing, I finally ended up with a new device in the deconz GUI (via VNC).  It appears to have four "Home Automation Level Control Switches". However it does not seem to be shown in the Phoscon web app, nor in the REST API.

Is it the same as the iCasa remote, or a separate device (manufacturer code is shown as 0x1224)? Am I doing something wrong in trying to add the device, or are changes to deconz needed to add support for it? (In that case I suppose a new device support issue is appropriate). Any suggestions are welcome.

I use deconz 2.05.80 running in a docker container.

[image]

Mimiix, 2021-01-14:
@spiff42 best is to open a new request :)

rweise, 2021-01-14:
i do not know if this is the right place to answer. I think there is a problem with the architektur of deconz or zigbee , i do not know. I own 3 different Remotes  of this kind. iCasa, Sunricher and RGBgenie. they are all one kind with the same Firmware 2.5.3_r20 only the Manufackturercode and modelcode are different. But first they where all complete different in iobroker that i use. in the meantime icasa and sunricher work the same way, but RGBgenie has a object for every single buttonpair and if i use them deconz will crash very fast and dirty. i would like to use the RGBgenie, because they have not the S label on the last row. So i open the remotes and changed the buttonrubber. i can not understand why it is not possible to tell deconz that RGBGenie remote or your ZGR904 is the same like sunricher or icasa. Unfortunately i wana send one of my 2 RGBgenie to a deconz programmer, but the post returned the parcel with unknown reason and then i decided to change only the rubber. Sometimes i try a new release of deconz, but with all last versions, my java scripts will not work anymore, so i change back. Since some days, i have a deconz test system, mainly for testing a combination of SSD and Raspbee or Conbee. But it make also a try with the latest Soft and Firmware, without any luck. RGBgenie crashes the deconz system. (And by the way, ssd and Rasbee or Rasbee 2 will not work, and Conbee only wit a 2m USB extension cable for 2 m between Pidesk with SSD and the conbee stick.

rweise, 2021-01-14:
And last but not least, you can find me on the discord deconz channel  and if you want to have a icasa from ersatzteilshop,de then find me there and we find a way to bring it on the way to denmark.