dresden-elektronik/deconz-rest-plugin #2649: 26350500 - 2.05.75 - Cannot pair Philips HUE Dimmer

WhistleMaster, 2020-04-04:
Hi,

I'm using Deconz  v2.05.75 with Conbee I on firmware 26350500 with Docker (marthoc image). The Conbee is attached to a NUC with an USB extension cable.

I'm trying to add two Philips HUE Dimmer (4 buttons) but I can't pair them through the Phoscon App. 

I go to Switches -> Add new switch -> Philips -> Dimmer Switch.
I press on the Setup button and release when the dimmer light turns green. 
Then, the light on the dimmer turns off but the dimmer is not added on Phoscon.

If I go to Deconz interface with VNC, I can see that a new device appears, but nothing more:

[image]
[image]
[image]

I tried to remove the battery from the dimmer, start again the pairing procedure, press on the 4 buttons to do a soft-reset, nothing helps.

Thanks for your help !

ebaauw, 2020-04-04:
The 0x01 endpoint is not shown.  Looks like deCONZ didn't read the _Simple descriptor_ for that endpoint.  Could you try to _Read Node Descriptor_ and _Read Simple Descriptor(s)_ and see whether the first endpoint appears?  Make sure to press a button on the dimmer every seconds or so, to force it to be awake.  If that doesn't help reset the dimmer (hold the reset button for 10 seconds), and try to re-pair it.

WhistleMaster, 2020-04-04:
I've seen that the devices were in the `zll.db` sqlite database with the "deleted" state (`deletedState`) in the `sensors` table and were also in the `devices`.

I shutdown Deconz, removed the entries from the tables, startup Deconz and tried to add the devices again.

I had to do what you suggested (_Read Node Descriptor and Read Simple Descriptor(s)_) and it worked.

I don't know if deleting the "deleted" devices from the database helped or not... @manup 

Thanks for your help !

mat1990dj, 2020-04-07:
It seems what I have but with different devices
[image]
Should I delete from zll.db?

WhistleMaster, 2020-04-07:
If you deleted the `zll.db`, you'll loose all your devices. You need to use `sqlite3` to delete only the rights entries.

mat1990dj, 2020-04-07:
Yes, I deleted the entries from zll.db and then started deconz, the ghost devices are gone but if I try to pair them again they come again in this strange format without name and not pairing

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

sunshineautomate, 2021-07-06:
I'm still experiencing this issue! Can it be reopened please?

Mimiix, 2021-07-07:
@sunshineautomate I rather have you post your issue on the forum. https://forum.phoscon.de :)