dresden-elektronik/deconz-rest-plugin #2650: 2.05.75 lights connectivity gone w/o change

nin9s, 2020-04-04:
//this may be related to another issue, however didnt find the it yet

all of a sudden the zigbee connectivity for lights stopped working without any config changes. I've already restored the VM which is running conbee2 but that didnt make a difference.
All sensors are working fine but everything which is categorized as "light" isnt.

debug log doenst show anything special apart from the following which keeps repeating:

```
12:01:14:854 no button map for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021
12:01:14:854 ZCL attribute report 0x0017880104E55F61 for cluster: 0x0001, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
12:01:20:107 Bind response success for 0x0017880104e55f61 ep: 0x02 cluster: 0xFC00
12:01:21:307 The daylight sensor seems to be configured with invalid values
12:01:31:306 The daylight sensor seems to be configured with invalid values
12:01:41:306 The daylight sensor seems to be configured with invalid values
12:01:51:306 The daylight sensor seems to be configured with invalid values
12:02:01:306 The daylight sensor seems to be configured with invalid values
12:02:11:306 The daylight sensor seems to be configured with invalid values
12:02:11:331 Current channel 11
12:02:11:343 Device TTL 2180 s flags: 0x7
12:02:21:306 The daylight sensor seems to be configured with invalid values
12:02:31:306 The daylight sensor seems to be configured with invalid values
12:02:41:306 The daylight sensor seems to be configured with invalid values
12:02:51:307 The daylight sensor seems to be configured with invalid values
12:03:01:307 The daylight sensor seems to be configured with invalid values
12:03:11:306 The daylight sensor seems to be configured with invalid values
12:03:11:329 Current channel 11
12:03:11:340 Device TTL 2120 s flags: 0x7
12:03:21:306 The daylight sensor seems to be configured with invalid values
12:03:31:306 The daylight sensor seems to be configured with invalid values
```
changing the channel also doesnt work.

How can I debug this issue? What causes this? Before restoring the vm there where "0xA7 No application ACK received" errors in log but these are gone currently.

Thanks for helping me out

kroxas187, 2020-04-07:
Having the same problem here. Sensors are all working fine, but anything characterized as a light drops out here and there. Sometimes they're gone for a few hours, other times days..

nin9s, 2020-04-07:
Figured sensors were not working too. They were not unavailable but didn't update anymore. Only way to solve this was to reset all devices. but tbh this really can't be a permanent solution.

pponce, 2020-04-10:
I'm having the same issue.
It happened to me after update to deCONZ_Rpi_0x26350500.
Thinking it might be related to losing the zigbee network config I tried the recommended fix from here
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1226

The only other change besides the firmware update of deconz was that  i also updated nodejs to 12.16.2 (from 12.16.1) but i don't  think that is related to this issue.

pponce, 2020-04-11:
Looks like i just had to wait a while. Not sure how long but after letting things "settle" for a bit all the lights were reachable again.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix