dresden-elektronik/deconz-rest-plugin #2651: Eurotronic Spirit Zigbee Setpoint / State Question

fenderle, 2020-04-04:
I have a question about the Eurotronic Spirit Zigbee:

When I set config.mode to "off" everything seems to work for a while, but after a few minutes config.heatsetpoint changes to 500. When I set config.mode back to "auto" it restores also the correct setpoint. However this throws several things off. E.g. the representation in Home Assistant shows 5.0C with mode set to off and it also throws off my NodeRED automations (because I expect the setpoint to stay at the actual configured value).

I have seen in the VNC GUI that the thermostat is actually sending that temperature change, so my hopes are low in getting this resolved somehow, but would it be possible to report the set setpoint rather than the invalid 500 setpoint?

ebaauw, 2020-04-04:
> because I expect the setpoint to stay at the actual configured value

That's a conversation you need to have with Eurotronic.  I don't think any-one has been able to get a response from them, though.

> I have seen in the VNC GUI that the thermostat is actually sending that temperature change,

Indeed, this is how the TRV seems to work: it changes the setpoint.  Same when setting mode to "on" for boost mode.

> would it be possible to report the set setpoint rather than the invalid 500 setpoint?

The TRV has changed the setpoint to 500; there's nothing invalid about the value.  It still remembers the previous setpoint somehow, and returns to that when canceling off mode (presumably this is meant for open window detection).  It doesn't report the previous setpoint.  I found that when switching from off (open window mode) to on (boost) or the other way round, the previous setpoint is lost, so I daren't guess the previous setpoint in the REST API plugin.

fenderle, 2020-04-04:
Thank you for clarifying this. Sadly thats what I expected. The firmware seems to have a lot of 'problems'.

closed 2020-04-04 by fenderle