dresden-elektronik/deconz-rest-plugin #2652: Eurotronic Spirit Zigbee Setpoint Reporting

fenderle, 2020-04-04:
When I manually change the setpoint on the valve using the +/- buttons it is not reported back through the REST API. From the VNC console it seems like the TRV updates attribute 0x0012 but not attribute 0x4003, the latter being the one which is most certainly reported back. Attribute 0x4003 is changing with a big delay while 0x0012 changes right away.

ebaauw, 2020-04-04:
Looks like the REST API plugin failed to setup attribute reporting properly.  Reporting should be disabled for 0x0012 (to conserve the battery, as this attribute is ignored by the plugin) and should be enabled for 0x4003.  

You can try to reset and re-pair the TRV, or setup the reporting manually in the GUI.  For the settings, see: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/ca72c46fd147dcf5c0b04304a96316d2c97ff477/bindings.cpp#L992-L1039

ebaauw, 2020-04-04:
Oh, while you're at it, better check the reporting configuration for the battery as well: cluster 0x0001.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/ca72c46fd147dcf5c0b04304a96316d2c97ff477/bindings.cpp#L1115-L1120

fenderle, 2020-04-04:
Jep - except for battery percentage and host flags all attributes were set to different values (I assume defaults). Now it works! Thanks!

closed 2020-04-04 by fenderle