dresden-elektronik/deconz-rest-plugin #2654: Could not add Light 1 to group. Group capacity of the device is reached.

paolotremadio, 2020-04-04:
Hi, 

I'm trying to group 4 new lights on deCONZ in a LightGroup but I'm getting 
> Group capacity of the device is reached.

What's the group capacity for the device?

ebaauw, 2020-04-04:
In ZigBee there's no object for groups; a group is just a multicast address, to which lights (and other devices) subscribe.  Each light maintains a table of the groups it has subscribed to.  This table has limited space, depending on the device (and probably the device firmware).  Hue lights can join up to 25 groups; other lights might support considerably fewer groups (as few as 8).  Note that the default group (for `/groups/0`) already occupies an entry.

To see how many groups a device has joined, and possibly how many entries are still available, open the _Cluster Info_ panel in the GUI and select the server (blue) _Groups_ cluster of the light.  Then issue a _Get group membership_ command.  This should return the _Group count_ (number of groups joined) and the _Capacity_ (number of entries still available).  Don't be overjoyed when that reports 255, it just means that there's at least 1 entry available.

paolotremadio, 2020-04-04:
I’ve learned something new today. 

What’s really strange is that I’ve just pared the light to deCONZ, I don’t see how it could be full already. The 4 lights are Philips GU10 white & ambience (without Bluetooth). 

Is there anything else I have to do after pairing to make it work?

ebaauw, 2020-04-04:
Maybe the REST API plugin hasn't yet read the _Groups_ cluster and defaults to thinking no entries are available?  If you start deCONZ with enough debug options, it logs the querying of the _Groups_ clusters of the devices with `/lights` resources.

paolotremadio, 2020-04-06:
Removed the light, resetted it, re-added it and now it works. 
I've written down the solution about forcing a read of the groups cluster.

Thanks!

closed 2020-04-06 by paolotremadio