dresden-elektronik/deconz-rest-plugin #2655: Q: Pair hue lights with ConBee2 or HUE bridge?

info-path, 2020-04-04:
Hello guys, this are more questions than an issue but I need to fully understand what to do. I spent a few hours searching but I'm still not sure about this.

I bought [ConBee2](https://phoscon.de/en/conbee2) and I have successfully installed and running [marthoc/deconz](https://github.com/marthoc/docker-deconz) container on Ubuntu, which is fully functional and listens on port 80. There was no problem of integrating into Home Assistant. 

I have many Philips hue lights connected to a hue bridge. I tried to pair them through the Phoscon app via ConBee2 and everything was successful. So far so good.

_Here's what I don't understand_

I assumed that it would be possible to add a _Phoscon-GW_ to [Philips Hue](https://play.google.com/store/apps/details?id=com.philips.lighting.hue2) app (Android) to control the hue bulbs connected to ConBee2 via Phoscon app. I succeeded in adding the _Phoscon-GW_ bridge to the Philips Hue app, but I immediately get a message about some necessary update, which always ends with some fail message. No further action is possible - update cannot be skipped. Note that firmware all connected bulbs is up to date. So, Is there a way how to connect _Phoscon-GW_ bride to hue app and avoid similar issues with the update request?

_Is there any solution?_

I don't insist on a hue bridge or Philips Hue app, but I actively use two entertainment areas for TV and PC and I have no idea, how them define without using Philips Hue app. I should mention that for hue entertainment I don't use the original [Philip Sync](https://www2.meethue.com/en-hk/entertainment/hue-sync) on my PC, but a third-party app - [Huestacean](https://huestacean.com/). And of course I'm addicted to entertainment areas :)

Maybe the only solution is connect all hue bulbs into the hue bridge and everything else like sensors and switches pair with Phoscon bridge. Is it?

I am sorry to bother you, but I would like to know which way is right.

Thanks for any clue.

wvuyk, 2020-04-05:
Here I have the same situation. For the Entertainment functions to work, the lights **_must_** be connected on the Philips Hue bridge. deCONZ does not support the entertainment api, so it wont work on deCONZ at all.
I ended up with having only the entertainment lights connected on the Philips Hue bridge. All other lights and sensors are on deCONZ here.

info-path, 2020-04-05:
Ok, thanks for reply. I thought it would happen with my hue bulbs :) I use them in different combinations of scenes, so it sounds logical to control them by one application/bridge.

Can you please confirm that original app Philips Hue for android cannot operate with Phoscon-GW? Or it is possible and only I have a problem with the issue that an update is required after connecting? I tried to connect the Phoscon-GW in Hue Essentials without any problem - everything works there.

rasssta, 2020-05-17:
I'm having the same update issue on iOS.

u20p17, 2020-06-01:
Will there be an update in the future that deconz is also supporting the hue entertainment api?

Mimiix, 2020-06-06:
We can only hope @u20p17 :( I'll close this issue for now. Open a featuure request issue if you want to:)

closed 2020-06-06 by Mimiix