dresden-elektronik/deconz-rest-plugin #2656: Xiaomi Mi Magic Cube in Homeassistant without state

vgisbert, 2020-04-05:
I have a Xiaomi Magic Cube MFKZQ01LM integrated through CONBEE II on Hassio (Homeassistant) and I can't see the status from HA. However, it is displayed in the deCONZ application.
The only variable that does carry over is the battery level.
It is the only zigbee device that does not work well for me.

Attached screenshots.

[image]

[image]

[image]

There are some similar but not the same issues.

Any ideas?

SwoopX, 2020-04-05:
Can you check if the corresponding sensors are exposed through the REST API? Check out https://dresden-elektronik.github.io/deconz-rest-doc/ for the details.

Kane610, 2020-04-05:
This follows the same mechanics as a remote in hass. Read integration documentation to understand how it works

vgisbert, 2020-04-05:
Thanks for answering, my architecture: INTEL NUC + Ubuntu 18.04.4 LTS + DOCKER + HASSIO (Home Assistant 0.107.7) + Home Assistant Add-on DECONZ and according to the information:
https://www.home-assistant.io/integrations/deconz/
I think the system already uses the REST API, although in the supported switches the XIAOMI MAGIC CUBE does not appear.

vgisbert, 2020-04-05:
Thank you all, I have already found the information.
In the document itself:
https://www.home-assistant.io/integrations/deconz/#remote-control-devices
There is a section dedicated to CUBE that says:

"Remote control devices
Remote controls (ZHASwitch category) will not be exposed as regular entities, but as events named deconz_event with a payload of id and event and in case of the Aqara Magic Cube also gesture. Id will be the device name from deCONZ and Event will be the momentary state of the switch. Gesture is used for some Aqara Magic Cube specific events like: flip 90 degrees, flip 180 degrees, clockwise and counter clockwise rotation. However, a sensor entity will be created that shows the battery level of the switch as reported by deCONZ, named sensor.device_name_battery_level."

It is clear that the only entity of this device is its battery "sensor.device_name_battery_level" and that the other data must be obtained through the available services: configure & deconz.device_refresh

There is nothing like knowing how to read and have patience. Thank you all.

closed 2020-04-05 by vgisbert