dresden-elektronik/deconz-rest-plugin #2657: Klik Aan Klik Uit ZTHS-100 zigbee binding problem

martinvanbuuren, 2020-04-05:
Hey all,

I'd like to switch completely over to home assistant for my home automation. One of the products I'm currently using to control my ventilation system is the Klik Aan Klik Uit (SEO: KAKU Trust) ZTHS-100 Temperatuur- en Luchtvochtigheidsensor (Eng: Temperature and Humidity sensor). At the current date, this product is not listed as a supported on deCONZ. That is why I was looking to gather information for a Device Support Request.

Unfortunally, I'm stuck. The Phoscon Windows 10 app can not find the sensor ("Failed to connect" after 3 minutes) even though the sensor is in learning mode, and hence, I cannot see it in deCONZ (although it seems to see a device with a battery, which I don't understand). I know the sensor is not broken because it connects perfectly with the Klik Aan Klik Uit control station ICS-2000. And on the other hand, other products I purchased from their Zigbee line do work perfectly with the conbee ii, such as the ZPIR-8000 motion detector and some Zigbee smart lights. Therefor I conclude that the conbee ii (and most likely the software) is not broken. 

So lately I've spend a couple of hours trying to and thinking of a reason or a solution, so far without any luck. Any insight is very appreciated.

Thanks!

Hartelijke groet,

~Martin

SwoopX, 2020-04-05:
The device is currently not supported so you need to raise a corresponding request. Please refer to the wiki for further details. Thanks.

martinvanbuuren, 2020-04-05:
Thanks for the respons. That is what I’m trying to explain: it cannot bind so I cannot file a request.

SwoopX, 2020-04-05:
If Phoscon does tell you "Failed to connect" does not necessarily mean that the pairing has not completed successfully. It basically tells you either that or that no sensors have been created. Therefore, check the deconz GUI for the device in question.

ebaauw, 2020-04-05:
It looks like an OEMed Heiman.  It should be able to join the network created by deCONZ.  Are you sure you factory-reset the device correctly?  How many routers and end-devices does your network currently have?

martinvanbuuren, 2020-04-06:
Thanks guys, that's it. I expected that initial pairing must be done though Phoscon (with an message telling that the device is not supported) and deCONZ was just to read the details out. The device it saw with the battery, as mentioned in the first post, was indeed the ZTHS-100. For future reference to others: you should click the right of the two gray buttons to access the details.

I'll make an overview of the sensor's Device Support Request. Thanks again!

PS. FYI as a side note: The motion sensor Klik Aan Klik Uit ZPIR-8000 shows up in deCONZ as an Adurolight:
- NWK: 0x0770
- Name: VMS_ADUROLIGHT
- Model: VMS_ADUROLIGHT
- Vendor: Trust International B.V.
- Version: v0.1.5
However, it is recognized as a motion sensor in Phoscon and it signals correctly, so I think only the naming feels off. Feel free to ask me for additional screenshots or information if you seems fit.

SwoopX, 2020-04-06:
Seems like this one is supported already. Apparently, there was a hickup on your first join attempt.

martinvanbuuren, 2020-04-06:
Let's keep it on a hickup. :-) The most relevant screenshots are displayed below, see the attached zip below for a full overview.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

All 16 screenshots are contained in the ZIP-file hereafter. Please notice that the orders of the files correspond with the clusters in the overview, when sorted on file name.
[ZTHS-100_deCONZ_All_Screenshots.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4437725/ZTHS-100_deCONZ_All_Screenshots.zip)

SwoopX, 2020-04-06:
Thanks. Should be enough to do the magic.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix