dresden-elektronik/deconz-rest-plugin #2659: Sunricher Wall Dimmer SR-ZG2835RAC

flopp999, 2020-04-06:
Any one know if this is working with Conbee 2 and Dresden?

https://www.sunricher.com/zigbee-series.html

https://www.m.nu/zigbee-inbyggnad/zigbee-vriddimmer

SwoopX, 2020-04-07:
Chances shouldn't be to bad. However, some amendments and testing may be required when we know, which clusters are available.

flopp999, 2020-04-07:
Here is a link to the PDF.
https://www.m.nu/Common/File/Download?guid=47823922-806D-4312-B474-573DC0D8DEDE

So if I buy one someone, can someone tell me how to help you to get what you need to include it to Dresden?

SwoopX, 2020-04-07:
Cool, thanks. Nothing extraordinary from the clusters perspective, so it should do quite alright. I can help you with that.

gejr, 2020-05-03:
Any updates here? I am also considering getting this one. Would love to know if it's possible to get it working with my conbee2/deconz.

SwoopX, 2020-05-03:
Someone must have the device before we even can do anything. We need to have the details as described in the wiki for a device support request and somebody willing to stick with the testing where required.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

joggs, 2020-08-05:
I've just installed one of these dimmers. After adding it to deconz as a light, I can see it in the phoscon app and can turn it on/off and also dim it. When it comes to the dimming, my lights are at full brightness until I get down to around 30 (of the max value in phoscon ui which is 100), so that could be improved. I'll post the details for the dimmer soon. It also pops up as a dimmable light in home assistant after a restart.

joggs, 2020-08-05:
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-08-05:
Electrical mesurement & simple metring ?
For your dimming problem, I think it s from the bulb, not the device ? Are you sure your bulb are dimmable too ?

alexdepalex, 2020-08-05:
I have the same device, rebranded by Robb shop (ROB_200-014-0), but it works perfect for me. I use it with a Philips DimTome LED spot, and that also has a weird dimming range, but with a one color led bulb, dimming looks fine. I also just checked the values for electrical measurement, and they're correct for the load I've connected to it.

Mimiix, 2020-08-05:
Based on @alexdepalex his comment and https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2829 (thanks @SwoopX ) This seems to be supported already.

If you have a issue with it, please join our discord or open a [User Question issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=User+Question&template=user-question.md&title=)

closed 2020-08-05 by Mimiix

flopp999, 2022-12-01:
I finally bought 3 of them now. I did get On/Off and could also dim but power and consumption didn't show up.
Built new JSON that looks like this. I take no responsible that it is correct but it is working for me :)

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Sunricher",
  "modelid": "HK-SL-RDIM-A",
  "product": "HK-SL-RDIM-A",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/hk-sl-rdim-a.json",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "description": "The measured consumption (in Wh).",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "default": 0
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Smanar, 2022-12-02:
@flopp999 

>but power and consumption didn't show up in HA.
>EDIT: added division for Power and Consumption

But it work or not ?

`"ep": 0,`

Try to use the endpoint 1.

BTW you are missing the reporting/binding for the power/tension/ect ....
Mean if you read the value using the GUI, it will update the value in the API, but you will not have automatic report.

You can compare with DDF for plug for cluster 0x0702 and 0x0b04

flopp999, 2022-12-02:
It works as it is in the JSON above.

I will try with your ideas during the weekend and report back here