dresden-elektronik/deconz-rest-plugin #2661: Phoscon WebUI Port changed to 80 after automatic update

iotathome, 2020-04-06:
The WebUI Port of my installation was 8088. A few days ago I tried to connect to Phoscon on my Raspberry 3, but couldn't connect. I found out that Phoscon was now listening on Port 80, which conflicted with another Webserver (Pi-Hole). After correcting the port in /lib/systemd/system/deconz.service and restarting the service everything was ok.

My current version is 2.5.74

TungstenE2, 2020-04-06:
Problem is existing for some time. Fix would be nice.

ebaauw, 2020-04-06:
Don’t change the service file, it will be overwritten with each update.  Use a drop-in file instead.  See `man systemd.unit`.

TungstenE2, 2020-04-06:
see also:

https://github.com/dresden-elektronik/phoscon-app-beta/issues/26

TungstenE2, 2020-04-06:
https://canox.net/2019/07/conbee-ii-einrichten/

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix