dresden-elektronik/deconz-rest-plugin #2662: Kadrilj Blinds not exit in Phoscon and no function

Maddin2020, 2020-04-06:
Hello, i want control my ikea Kadrilj blinds with the conbee 2 stick and openhab. Now i can connect the blinds in the rest Api but i see no reaction in openhab and in the rest Api if i control the blinds. In the phoscon app now i see no blinds. I have read in another issues that Deconz want to create a new window for the blinds and betwen the blinds will be filtered. Now i can´t see where the fault is. (openhab Binding or the Phoscon App) because i don´t can control the blind directly from the Phoscon App. When Deconz implemate the changes for the Ikea blinds ? Can everybody give me a example for Openhab and the Ikea Blinds ? Thanks a lot.

SwoopX, 2020-04-07:
The Phoscon page for the blinds will arrive with the next release, if I'm not mistaken. Please note that Phoscon is not open source and not part of this repo.

For the API topic, if I understand you correctly, you paired via deconz GUI? Use Phoscon for that and eventually repeat (have the blinds physically reset) if it doesn't work out on the first attempt.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix