dresden-elektronik/deconz-rest-plugin #2663: Marthoc/deConz 2.05.75 - RaspBee2 - rPi 4b - channel change not working

gidig, 2020-04-07:
Since installing a RaspBeeII in an RPI4b / Rev 1.1 I have had problems adding devices (as mentioned in many other threads).

A configuration with SD card, Ethernet, Wi-Fi and BT off gave the first configuration success.
For further tests I tried to change the ZigBee channel.
I even managed that once (11-> 20) but since then the channel has stubbornly got stuck at 20.
With the Phoscon app, a change to channel 11 stays for about 30 seconds before it jump's back to 20.

The deConz app (Network Settings) and the db (config2/zigbeechannel) shows channel 11 as active
but the phoscon page still shows channel 20 even after restarting deconz.
The log file is flooded with channel change messages like this.

deconz         | 10:56:15:739 Current channel 20
deconz         | 10:56:15:761 Device TTL 6480 s flags: 0x7
deconz         | 10:56:25:694 channel 20 does not match channel mask 0x00000800 (TODO)
deconz         | 10:56:25:694 channel is 20 but should be 11, start channel change
deconz         | 10:56:25:694 Skip automatic channel change, TODO warn user
deconz         | 10:56:35:694 channel 20 does not match channel mask 0x00000800 (TODO)
deconz         | 10:56:35:694 channel is 20 but should be 11, start channel change
deconz         | 10:56:35:694 Skip automatic channel change, TODO warn user
deconz         | 10:56:45:693 channel 20 does not match channel mask 0x00000800 (TODO)
deconz         | 10:56:45:693 channel is 20 but should be 11, start channel change
deconz         | 10:56:45:693 Skip automatic channel change, TODO warn user
deconz         | 10:56:55:694 channel 20 does not match channel mask 0x00000800 (TODO)
deconz         | 10:56:55:694 channel is 20 but should be 11, start channel change
deconz         | 10:56:55:694 Skip automatic channel change, TODO warn user
deconz         | 10:57:05:694 channel 20 does not match channel mask 0x00000800 (TODO)
deconz         | 10:57:05:694 channel is 20 but should be 11, start channel change
deconz         | 10:57:05:694 Skip automatic channel change, TODO warn user
deconz         | 10:57:15:694 channel 20 does not match channel mask 0x00000800 (TODO)
deconz         | 10:57:15:694 channel is 20 but should be 11, start channel change
deconz         | 10:57:15:694 Skip automatic channel change, TODO warn user
deconz         | 10:57:15:738 Current channel 20
deconz         | 10:57:15:760 Device TTL 6420 s flags: 0x7

In this state it also doesnt pair with any new resetted device (which worked before and on RaspBee I).

MrHollowPS, 2020-05-23:
Out of curiosity what firmware version are you on, i just encountered this issue with a docker installation and i cannot seem to switch back from channel 20 to anything else :(

MrHollowPS, 2020-05-23:
Managed to fix it by flashing an OLDER FIRMWARE changing the channel back to 11 and resetting the gateway.

Now I am flashing the new firmware back $ GCFFlasher_internal -d /dev/ttyACM1 -t 60 -f deCONZ_ConBeeII_0x26580700.bin.GCF

Mimiix, 2020-06-12:
This seems to be fixed!

closed 2020-06-12 by Mimiix

tobycheese, 2020-10-20:
Thanks. That also worked for raspbee premium, downgraded to deCONZ_Rpi_0x26340500.bin.GCF, changed the channel via VNC I think (could also have been via REST) then upgraded firmware back up to deCONZ_Rpi_0x26390500.bin.GCF. Hat to restart the gateway via VNC to see the effect in the webapp.

pboschi, 2021-01-04:
> This seems to be fixed!

This doesnt't seem to be fixed if a firmware downgrade is needed in order to change the channel. 26350500 is OK on Raspbian, 26390500 is not.

jwrona, 2021-03-10:
I can confirm that this is not fixed. Cannot change the channel on firmware 26680700.

Mimiix, 2021-03-10:
@pboschi @jwrona Can either of you open a new issue? This issue doesn't use the template. Me re-opening is getting it closed instantly again.

jwrona, 2021-03-10:
> @pboschi @jwrona Can either of you open a new issue? This issue doesn't use the template. Me re-opening is getting it closed instantly again.

It's ironic, but when I was collecting logs for the new issue, the channel change actually worked. Without me changing anything. I'll open the issue when I'll be able to reproduce the problem again. ConBee is truly a peculiar device.