dresden-elektronik/deconz-rest-plugin #2664: Firmware shows latest even though there are newer

mat1990dj, 2020-04-07:
I've searched to see if this issue is already reported by didn't find anything.

My phoscon is showing the following:
[image]

Even though I've seen that there is a new firmware on http://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D

Is it working as expected?

SwoopX, 2020-04-07:
If I'm not mistaken, the newer firmwares are to be considered beta versions (at least for Conbee II). Not sure how the update check works and if it is based on any functionality from the deconz REST plugin, which can be checked.

Btw, Phoscon is closed source, before you ask.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix