dresden-elektronik/deconz-rest-plugin #2665: Vimar New Radiofrequency line that support both Zigbee and BT 5

cafedo, 2020-04-07:
Hi everyone,
do you know if the the new Vimar Radiofrequency line based both on Zigbee and BT 5 ( https://www.vimar.com/en/int/your-wiring-system-becomes-connected-15482132.html) is supported?

In particular I would like to know if these devices can be managed by deconz:
- switch: https://www.vimar.com/en/int/catalog/product/index/code/14592.0
- connected actuator for power: https://www.vimar.com/en/int/catalog/product/index/code/14593

Thanks in advance.

SwoopX, 2020-04-07:
In theory, yes. We can probably include support but as far as I can tell, they don't provide a detailed technical reference on the used clusters and enpoints (at least not publicly or I've overlooked it). You may want to reach out to their support and ask for it.

In case we don't have that documentation, somebody must buy it, issue a new device support request and stay with it for testing, which eventually can be rather technical. Even better would be to also have the corresponding gateway and sniff the traffic.

cafedo, 2020-04-07:
Ok, I’ve bought a switch to test; when it arrives, I’ll post a new device support request including all the info that I can track. Thanks.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix