dresden-elektronik/deconz-rest-plugin #2666: Problems removing and pairing devices

mat1990dj, 2020-04-07:
With latest .75 deconz and plugin master as of today, I have problems pairing with AQARA OPPLE buttons, and IKEA roller blind window cover.

When I pair them they will appear without name, but with 0X... name

If I try read simple descriptor it just unselects. I have a hard time removing again the devices, I end up removing even from zll.db.

[image]
This is an example of the four devices, when I pair them they have the link line, but name stays the same.

Is there a known issue with latest code? What version should I use? Can I try anything else?

Thank you

SwoopX, 2020-04-07:
From the screenshot you provided, the look like orphan devices. Don't seem to have a route to the coordinator and I would expect the roller blinds are mains powered?

However, from a pairing perspective, it looks like everything's gone alright, at least previously. So what's the pairing issue?

> If I try read simple descriptor it just unselects. I have a hard time removing again the devices, I end up removing even from zll.db.

The "unselection" you describe is "normal" behaviour, so I'm sure the simple descriptors have been read. What was the reason for reading them again? And did did you try to delete the devices on DB level?

As a side note: the blinds do not show up in Phoscon (not yet).

ebaauw, 2020-04-07:
> I would expect the roller blinds are mains powered?

The IKEA FYRTUR and KADRILJ smart blinds are battery-powered end devices.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-524488900.

mat1990dj, 2020-04-08:
Maybe I'm mistaken from the GUI representation?
When a device is paired to the network it should have a route (colour line) right?
So this is a paired device:
[image]

If it doesn't have a route, but has the clusters dots, it means that it has read the descriptor and also knows the name, but for some reason that decvice is not paired to the network (and I'm not able to pair it)
[image]

And if it only has one dot, I understood that it hasn't been able to read the descriptor, and doesn't know the name, and it's not paired (and i'm not able to pair it)
[image]

So in my case I have three different devices that won't pair to the RaspBee nor be deleted, and I would like to have them paired.

ebaauw, 2020-04-08:
> When a device is paired to the network it should have a route (colour line) right?

No.  The line indicates that the device is in its parent’s neighbour table, _and_ that deCONZ has recently read that table.  It’s quite common for end devices to show without a line. 

If the node shows in the GUI, it has joined the network, at least at some time.  If the device has left the network, a stale node might still show.  If the right dot shows, deCONZ has at least read one  simple descriptor, with an endpoint and clusters.  There’s a descriptor per endpoint.  The name on the node changes when the REST API plugin creates the resource(s) for the devices.

alex3305, 2020-04-10:
I had the same issue with my blinds as described in this comment in the linked thread https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-595301639 by @ebaauw.

I finally fixed my problem with some help from others (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1121#issuecomment-595931870)

> * Put Phoscon in pairing mode
> * Put the FYRTUR / KADRILJ in pairing mode (short press both buttons)
> * Long press both buttons on the FYRTUR / KADRILJ until paired with deCONZ

This process isn't described in the IKEA manual, but it has worked wonders for me. Maybe you can try this procedure yourself?

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix