dresden-elektronik/deconz-rest-plugin #2667: {Device Support Request] Sengled E13-N11WA - Smart LED with Motion Sensor PAR38 Bulb

stevewull, 2020-04-08:
deCONZ successfully recognizes and controls the light bulb component. However, it does not recognize the LUX meter or the motion sensor.  Please add support for these sensors.

[image]

Node info:
[image]

Basic cluster info:
[image]

IAS Zone:
[image]

Simple metering zone:
[image]

SwoopX, 2020-04-08:
Thanks, done.

stevewull, 2020-04-08:
What should I be looking for in the deCONZ app, API or Phoscon interface to check this?  Will it show up as a sensor in Phoscon, or only appear in the api?

SwoopX, 2020-04-08:
Just download and compile my repo here https://github.com/SwoopX/deconz-rest-plugin . From there, I pull my changes over and the changes are already included. Presumably, the motion sensor should work now. And you will have an additional consumption sensor.

stevewull, 2020-04-08:
It will show up as two nodes (Dimmable Light and Occupancy Sensor) with different MAC addresses?

SwoopX, 2020-04-09:
Hu? May I see that please?

Just to be clear on what you see via the API: The uniqueID consists of the MAC address, the endpoint and the cluster (for sensors) `00:17:88:01:04:f4:4e:17-02-fc00` seperated by a dash. It's that what you see?

How about the motion sensor  this time, did it enroll automatically? I would have also expected a  consumption sensor to show up...

ebaauw, 2020-04-09:
> It will show up as two nodes (Dimmable Light and Occupancy Sensor) with different MAC addresses?

What makes you think that?  See the screenshot above: no.  The GUI shows a node per Zigbee device, with a unique mac address.  Typically that’s a physical device, although some Hue luminaries contain multiple independent Zigbee modules, each with their own mac address.  The REST API shows a resource per endpoint (for `/lights`) or per cluster (for `/sensors`).  As @SwoopX explains, the `uniqueid` links the resource to the node.  It’s up to API clients (like Phoscon) how they display the resources: as separate entities, or as one.

stevewull, 2020-04-09:
I understand that the MAC address is the unique ID.  I asked because a new node just appeared on my network. It has a mac address that I do not recognize and is labeled an occupancy sensor. Setting that aside, I'll pull from your repo and recompile. For the PAR38 bulbs, I still only see the light portion in the API.

BUT, let me repull from you repo, recompile, and try again.

SwoopX, 2020-04-09:
I'd recommend to share some screensots or past the API output here. Maybe we can offer an explanation if we see it.

stevewull, 2020-04-10:
how do I clear the deCONZ database so that I am starting fresh to load all of the nodes from the network?

SwoopX, 2020-04-10:
You can do a `sudo rm -rf .local/share/dresden-elektronik/`. Make sure you've stopped deconz before with `sudo systemctl stop deconz-gui`.

stevewull, 2020-04-13:
The back and forth of using the same Zigbee network for both test and production has disrupted my production environment. I need to fix/stabilize, probably put half the devices into pairing mode and rejion. Then I can return to testing your latest changes.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

stevewull, 2020-10-13:
Re-opening this issue as it was not fully resolved when I stepped away. I recently installed deConz version 2.05.81.

The cluster and node info is the same as the screen shots above.

In Phoscon, these appear as lights. Product is labeled "Moiton Sensor." They do not appear in the sensors section.

[image]

In the API, it shows up in the lights section only:

    "12": {
      "etag": "a7409a45a0911691d11b1fd67b777d63",
      "hascolor": false,
      "lastannounced": null,
      "lastseen": "2020-10-13T05:44Z",
      "manufacturername": "sengled",
      "modelid": "E13-N11",
      "name": "Driveway Left Flood Bulb",
      "state": {
        "alert": "none",
        "bri": 254,
        "on": false,
        "reachable": true
      },
      "swversion": "0000001A",
      "type": "Dimmable light",
      "uniqueid": "b0:ce:18:14:03:1a:05:e1-01"
    },

stevewull, 2020-10-13:
The bulb works well as a controllable light.  

It has built-in light and motion sensors. We were trying to read them.

SwoopX, 2020-10-13:
You are aware that you should also have at least one additional sensor? what does `/sensors` say?

stevewull, 2020-10-13:
It was not appearing in the sensors.  So, I repaired the two Bulbs.  They now appear in sensors section of the API. A few notes:

- One bulb appears twice, Types ZHAPresence and ZHAComsumption. The presence detection works. 
- Other bulb appears once: Type ZHAComsumption. I tried repairing again and still no ZHAPresence entry in API.

I don't see the LUX sensor for either bulb. Nor can I tell which cluster contains it. Note on my first image above, there is an unknown cluster.

  "27": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "4d574aa6271e50872d040c773ad0e0cc",
    "lastseen": "2020-10-13T15:14Z",
    "manufacturername": "sengled",
    "modelid": "E13-N11",
    "name": "Consumption 27",
    "state": {
      "consumption": 0,
      "lastupdated": "2020-10-13T15:14:12.451",
      "power": 135
    },
    "type": "ZHAConsumption",
    "uniqueid": "b0:ce:18:14:03:1a:0c:04-01-0702"
  },
  "28": {
    "config": {
      "duration": 60,
      "on": true,
      "pending": [],
      "reachable": true
    },
    "ep": 1,
    "etag": "e1a8c818d0ac5e9089d97afec169ba44",
    "lastseen": "2020-10-13T15:13Z",
    "manufacturername": "sengled",
    "modelid": "E13-N11",
    "name": "Presence 28",
    "state": {
      "lastupdated": "2020-10-13T15:13:31.990",
      "lowbattery": false,
      "presence": false,
      "tampered": false
    },
    "type": "ZHAPresence",
    "uniqueid": "b0:ce:18:14:03:1a:05:e1-01-0500"
  },
  "29": {
    "config": {
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "aeecd90d5f58d0102ab03ee5f8e45ceb",
    "lastseen": "2020-10-13T15:13Z",
    "manufacturername": "sengled",
    "modelid": "E13-N11",
    "name": "Consumption 29",
    "state": {
      "consumption": 3,
      "lastupdated": "2020-10-13T15:13:08.066",
      "power": 135
    },
    "type": "ZHAConsumption",
    "uniqueid": "b0:ce:18:14:03:1a:05:e1-01-0702"
  },

lordlala, 2021-07-28:
I would love to see motion get implemented. My understanding is the ZHA integration built into HA does this.

I currently have 4 of these flood lights scattered around my house. Motion would be amazing...

stevewull, 2021-07-28:
I'd like the motion implemented too. Never got it working.