dresden-elektronik/deconz-rest-plugin #2669: Heiman CO sensor not working

cosmo84, 2020-04-08:
Hi all,

I bought a Heiman gas sensor and a CO sensor. The gas sensor is working fine, also joining it was no issue. With the CO sensor I don't move forward. It seems it is not completely joined, since I can't even access the cluster info to enroll it manually by specifying the IAS_CIE_Address. 

[image]

I am playing around for some time now, but everytime I delete the node and rejoin it the result is the same. 

Best regards,

Stephan

SwoopX, 2020-04-08:
Have you physically reset the device before conducting another join? Make also sure you join via Phoscon.

cosmo84, 2020-04-08:
If you mean with physically reset taking off the battery for some minutes then the answer is yes. 
I use phoscon for joining, but it does not work. When I press the button nothing happens in phoscon, but the sensor appears in the deconz GUI as shown in the screenshot.

SwoopX, 2020-04-08:
Pulling the batteries off not necessarily means the device gets resetted. Isn't there any smal physical button or tiny hole where you can put a needle in? What does the manual say about a reset?

Another thing you could try meanwhile is put Phoscon in sensor seach and read the simple descriptors again.

cosmo84, 2020-04-09:
Yes, there is this tiny hole that I press. When the LED flashes slowly for some seconds, the device left the network. When it flashes fast it joined a network. This is besides pulling batteries the only thing I can do I guess. 
I started a search for the simple descriptors, but nothing changes. When I do so, the left signal of the sensor is blinking red instead of blue.

cosmo84, 2020-04-09:
A maybe interesting fact: When I click to read the node descriptors, the led in deconz is blinking blue, so it seems to work, but in fact nothing changes. E.g. the manufactorer code is still 0x0000.

SwoopX, 2020-04-09:
Can you please provide a screenshot of the node info panel? Feels indeed that something went wrong. Which model is that sensor?

cosmo84, 2020-04-09:
The sensor is a Heiman HS1CA-E smart carbon monoxide sensor.

[image]

SwoopX, 2020-04-09:
Hm, the node descriptor is not right. Have you used Phoscon for pairing? Have you also tried to bring the device closely to the coordinator?

cosmo84, 2020-04-09:
So, I did again a delete of the node and started the pairing process in Phoscon. Still phoscon does not detect the sensor, but it appears in the deconz GUI. I was now holding it just 2 cm away from the raspbee module, and now at least the node descriptor looks correct. Still not more than that...

[image]

SwoopX, 2020-04-09:
Much better. Now again the trick with re-reading the simple descriptors while the sensor search is active in Phoscon. Strange that your device is quite stubborn...

cosmo84, 2020-04-09:
The search in Phoscon is running, but still does not find the sensor. Reading the simple descriptors has still no effect at all. Very curious...

SwoopX, 2020-04-09:
You still have only one bullet on the right and not two?

cosmo84, 2020-04-09:
Yes, just one.

SwoopX, 2020-04-09:
Have you tried pushing that tiny hole just a second before reading the simple descriptors? The device must be awake.

cosmo84, 2020-04-09:
Yes I pushed it several times, and on top of that I even used the test button once, but no success.

ebaauw, 2020-04-09:
If memory serves, the previous model Heiman CO sensor would only wake after holding the test button for a few seconds.  Be sure to cover your ears...

cosmo84, 2020-04-09:
Pushing the test button changed the behaviour, that the LED in deconz GUI is blinking green, but no change when reading the simple descriptors. Also phoscon is still not finding the sensor and no second button on the right side.

cosmo84, 2020-04-10:
Guess what...today I wanted to play around a little bit more, opened deconz GUI and saw the second bullet on the right. I started sensor discovery and enrolled it in the GUI while pressing the test button. All done and working. The question of the day is now, what triggered this self-repair. It was just time passing from my last post till now, I did not touch the system, especially no reboots etc.
That makes the solution as strange as the problem itself. Many thanks to @SwoopX and @ebaauw for your support!!!!

closed 2020-04-10 by cosmo84