dresden-elektronik/deconz-rest-plugin #2670: How to remove old Conbee stick?

TungstenE2, 2020-04-08:
Hi all,

I switched for Conbee I to Conbee II stick.
How to remove old Conbee from Deconz and Phoscon?

thx

[image]
[image]

Smanar, 2020-04-08:
I m interested too ^^.

How are you done the migration ? with backup/restore ?

jensflorian, 2020-04-11:
I‘d look into the zll.db database file, make a backup copy and try to delete the records for Conbee I

ebaauw, 2020-04-11:
That won’t do you any good, as this info doesn’t come from the database.  You’re seeing the deCONZ instances returned by the appspot server on discovery, before Phoscon has established an API connection to any instance.  The old entry should disappear in half an hour, unless its cached by your browser.

To delete the old coordinator node from the GUI, select it and press Del.

TungstenE2, 2020-04-11:
Just deleting the coordinator did not do the trick. It reappears after restarting deCONZ. Any further ideas?

ebaauw, 2020-04-11:
Might try and delete it from the `devices`table.

TungstenE2, 2020-04-12:
how to delete from device table? any instructions on this?

jensflorian, 2020-04-12:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/833#issuecomment-427970554 - I recommend sqlitebrowser (https://sqlitebrowser.org). The database is located in 

> .local/share/dresden-elektronik/deCONZ

Always backup your database first, before manipulating it. You might also consider deleting the entries in `device_gui` table as well.

TungstenE2, 2020-04-12:
so the issue is more than 2 years old and still not fixed?
old devices keep reapearing.

TheWizz, 2020-04-14:
Yes, I see the same thing. While you can delete old CONBEE in DECONZ, it re-appears after next re-start. Even though you CAN probably delete it by hacking the database file, that should not be required since the UI does support deleting it (but doesn't apparently propagate that deletion to the database). So  this is a bug needs to be fixed.

-JM

Mimiix, 2020-06-05:
Is this still a issue?

MattL0, 2020-06-26:
YEs @manup

MattL0, 2020-06-27:
I was not able to see it in the database.. Going there show my old conbee : https://phoscon.de/discover

[{"id":"00212EFFFF022CD5","internalipaddress":"192.168.0.100","macaddress":"xxxxxxxxxxxxxxxxx","internalport":xx,"name":"Conbee I","publicipaddress":"xxxxxxxxxxxx"}]

MattL0, 2020-06-28:
How can this be resolved?

MattL0, 2020-07-19:
> I was not able to see it in the database.. Going there shows my old conbee : https://phoscon.de/discover
> 
> [{"id":"00212EFFFF022CD5","internalipaddress":"192.168.0.100","macaddress":"xxxxxxxxxxxxxxxxx","internalport":xx,"name":"Conbee I","publicipaddress":"xxxxxxxxxxxx"}]
> 
> still there when I remove the database completly. I think this is bugged into the conbee 2 itself when I cloned it from conbee 1 last year

Any idea? @manup? I completly reformated my pc. Still the save

Smanar, 2020-07-19:
I know you have reformated the PC, but have you make the try on another machine but without playing with datatase, on a clean install ?

When you make a backup there is too a file called sessions.sessions inside, you can take a look if you find the mac adress of the old conbee inside, if you don't find the mac adress on the database.

You haven't previous backup ?

TungstenE2, 2020-12-21:
any news on this? No matter how often I delete old Conbee, it reappears after reboot.

Even updated to 2.08.00 raspbian-stretch-beta latest.

[image]

Mimiix, 2020-12-21:
I've asked @manup to check it ^^:) Hope he replies soon!

SwoopX, 2020-12-21:
This can currently only be done by deleting all traces in the database.

TungstenE2, 2020-12-21:
so this is a bug in deconz for the delete option. Please have this fixed. Issue is open for 8 months and was even reported before. 

thx a lot

Mimiix, 2020-12-21:
> so this is a bug in deconz for the delete option. Please have this fixed. Issue is open for 8 months and was even reported before.
> 
> thx a lot

I've created a internal issue on it with high priority for Manuel. 

Don't worry, I'll get him to reply.

manup, 2020-12-21:
Currently the old coordinator node can only be deleted in the `zll.db` file and here the `devices` table, for example with SQLiteBrowser.

The delete node via GUI function is on the todo list to correctly delete everything related to a device.

Note having multiple coordinator notes doesn't cause harm other than looking odd.

Here is my coordinator graveyard:
[image]

TungstenE2, 2020-12-22:
thx for your feedback. It is good to hear that there is no harm, but from user perspective this is very iritating that a deleted node is reappearing after a reboot. So there is no much trust in the software, what else might be wrong.

eg I have a node which I renamed a couple of times, and the old name is reappearing after a reboot as well.
Is this a related issue, as the change in DB is not persitet?

[image]

TungstenE2, 2021-01-04:
Dear @manup thx for the fix in 2.9.0 for this issue.

May I ask how this was fixed?

In the change log you state 'Hide old inactive ccordinator nodes' rather than 'Remove old inactive ccordinator nodes'.

So am I right that you just do not dispaly inactive nodes anymore, but the notes still remain undeleted in the DB?
So the issue it self is not fixed, you just do not show it anymore?

[image]

manup, 2021-01-04:
The change is related to just view the active coordinator at a time (since this is always true).

Note: Coordinators shouldn't be deleted in the database automatically, there are quite some setups out there, including mine, where different ConBee / RaspBee units are used over time. That's not an issue since the inactive entries do nothing.

The issue still open is the manual deletion of nodes and related resources via the context menu or Delete short-cut.