dresden-elektronik/deconz-rest-plugin #2671: WebUI doesnt detect devices that the Desktop app detects

mike391, 2020-04-08:
I tried to pair 2 Xiaomi mija door sensors using the webapp.  I tried to add them using Sensors->Add Sensor-> Other.   The webui was never able to detect these sensors though.  However in the Desktop UI, both devices appeared, however I am unable to set their 'Name', 'Model' or change any settings for them. 
 Furthermore they do not show up in the WebUI under sensors.

[image]

[image]

SwoopX, 2020-04-08:
The pairing has completed, but the corresponsing sensors have not been created. You can try to read the simple descriptors once again (click on the right bullet of the sensor which gives you a menu) or starting a new sensor search in Phoscon, but make sure the sensor is "awake". You can wake it up by moving the magnet around in front of the sensor so that it should trigger. This must all happen within a couple of seconds for those devices.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix