dresden-elektronik/deconz-rest-plugin #2672: Remove Tampered Attribute form Osram Smart+ Motion Sensor

peer69, 2020-04-08:
Osram Smart+ Motion Sensor seems to be constantly reporting "Tampered"-Attribute as true. Any way to disable/remove this attribute? 
It's quite annoying when using homebridge-hue because this will show an error in the status. 

´´´{
  "config": {
    "battery": 80,
    "duration": 60,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "d4a62dc71b7dfc1ed9e84c402ef82888",
  "manufacturername": "CentraLite",
  "modelid": "Motion Sensor-A",
  "name": "Bewegungsmelder Flur OG",
  "state": {
    "lastupdated": "2020-04-08T17:48:54",
    "lowbattery": false,
    "presence": false,
    "tampered": true
  },
  "swversion": "����������������",
  "type": "ZHAPresence",
  "uniqueid": "00:0d:6f:00:0f:bf:ab:76-01-0500"
}´´´

jensflorian, 2020-04-11:
Same here. See also The comments after device integration  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/881#issuecomment-596836233

As the reported attribute is useless, disabling this attribute would be great. 

@ebaauw : Or is there a possibility to deactivate this in homebridge-hue without blacklisting the whole device?

ebaauw, 2020-04-11:
Does the device have tamper detection?  Some sensors have a little switch to check that they’re connected to the base plate.  Can you please show a screenshot of the sensor’s _IAS Zone_ cluster, after reading all attributes?  Particularly interested in the _Zone Status_.

> Or is there a possibility to deactivate this in homebridge-hue without blacklisting the whole device?

No.  If the device reports Tamper in error, this should be fixed in the device firmware.  Fat chance with Osram/Ledvance stepping out of this business.  If we create workarounds for that error, it should be in deCONZ, so all API clients benefit.

jensflorian, 2020-04-11:
The device has a reset button. Once you have touched it remains in the tampered state. Others have also noticed that there is no true tamper check, see https://github.com/Koenkk/zigbee2mqtt/issues/507#issuecomment-443433195

Attached is the requested screenshot from the IAS cluster:

[image]

[image]

Do you need any other clusters?

ebaauw, 2020-04-11:
So it reports that it's capable of Tamper detection, but it incorrectly sets this bit in the zone status.  I think we would need to whitelist the device to ingore the _Tamper_ bit and not to expose`state.tamper`?  Could you please show a screenshot of the _Basic_ cluster (after reading the attributes), to see how to whitelist the device?

jensflorian, 2020-04-11:
It reports as "Motion Sensor-A" from CentraLite.

This is the Basic cluster from the IAS Zone:
[image]

There is another in the Home Automation Occupancy Sensor cluster with identical values.

It looks like the manufacturer also populated the Date Code value with non standard ASCII characters, hence the replacement symbols (https://en.wikipedia.org/wiki/Specials_(Unicode_block)).

ebaauw, 2020-04-11:
Above commit should fix this.  I cannot test, as I don't have the sensor myself.  Looking at the code, the tamper bit in the status report should simply be ignored, when `state.tampered` isn't present.

Also removed `state.lowbattery`, since the sensor doesn't support that.

There was an inconsistency in the code, that these attributes weren't exposed for the Samjin `button` and `water` sensors on pairing, but are on restore from the database.  Fixed that as well.

jensflorian, 2020-04-12:
Thank you very much for the commit. Unfortunately I cannot test this against V2_05_75 
because of issue #2563 breaking my system - I'm still waiting for openhab 2.5.4 release.

Maybe @peer69 who started this thread could do this?

peer69, 2020-04-12:
Thanks for the effort, guys. Awesome! 
How would I actually test this The safest way? Do I have to build the addon from source or could I just replace files in the filesystem which I could easily restore? 
With my kid staying at home right now and my wife working at the hospital I am not really open to break anything on purpose right now.

udovic2100, 2020-04-15:
Thank you very much @ebaauw for the fix. How can I update my Deconz rest plugin?

ebaauw, 2020-04-15:
See the README how to compile and install the plugin, or wait for the next release.

udovic2100, 2020-04-15:
I compiled and installed the plugin but the tampered-flag remains. I find it interesting that it works on the Homebridge accessories tab but not in Apple HomeKit. @ebaauw

ebaauw, 2020-04-15:
Did you restart Homebridge after restarting deCONZ?

udovic2100, 2020-04-15:
Yes I did. And I need to correct myself... It is also working in HomeKit but the Icon shows "Triggered" although it is actually not triggered. So I guess the initial problem remains?! Thanks for your quick help though. Let's see what we can do about the bug.

ebaauw, 2020-04-15:
That’s a different issue.  Did you double-check what the API reports?  And the deCONZ GUI?

udovic2100, 2020-04-17:
I had a look at the raw output of the deCONZ REST-API using a REST client. I compiled the new file (libde_rest_plugin.so) according to your README and replaced the old one. I also restarted the server. The output of my REST client after the GET request looks like this:

` "9": {
    "config": {
      "battery": 90,
      "duration": 122,
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "8b5ec5b4c4727c9c10dc93b9902c525c",
    "lastseen": "2020-04-17T07:49:45.906",
    "manufacturername": "CentraLite",
    "modelid": "Motion Sensor-A",
    "name": "Motion Sensor",
    "state": {
      "lastupdated": "2020-04-17T07:49:45.906",
      "lowbattery": false,
      "presence": false,
      "tampered": true
    },
    "swversion": "����������������",
    "type": "ZHAPresence",
    "uniqueid": "00:0d:6f:00:0f:7f:65:e1-01-0500"
  }`

So in my opinion the new REST plugin is still forwarding "tampered": true. Maybe I used the wrong version to compile the plugin? Thanks for your help in advance!

ebaauw, 2020-04-17:
Did you compile from my fork?  The corresponding PR hasn't been merged yet.

udovic2100, 2020-04-17:
Thank you very much for your help @ebaauw! I finally got it to work. I really compiled from the wrong fork because I was just following the instructions in the README without using my brain. ;)

peer69, 2020-05-23:
As PR has been merged I suspect this should have been included in .76? Still getting the error in status.

closed 2020-05-23 by peer69