dresden-elektronik/deconz-rest-plugin #2673: Motion Sensor „from Desk to Dawn“ didn‘t work in Phoscon

bruesche, 2020-04-08:
Hi there,

Did a great error: I use the deconz / Phoscon with the „Hue Essentials“ App. In the Phonscon Web App I used the Motion Sensor setup with „from dusk Till dawn“. Everything works fine. Now I find Settings in „hue essentials“ for geografic Position - Offset and changed the setting to get an earlier Start off the light at sun down. Now nothing works as before I changed it with motion control. Only the option “alway” works now with motion sensors. 

What can I do to get an working solution with “from dusk till dawn” for my Motion Sensor setup? Is there any solution to reset the time for “from dusk till dawn” without resetting the whole gateway?

Hope anybody can help...

Regards
Thorsten

ebaauw, 2020-04-08:
The obvious answer is to restore a backup of the database from before changing the Daylight setting in Hue Essentials.

If you don't have such a backup: double-check the Daylight sensor settings in the database, or simply set them again from the API.  The deCONZ API and Hue API use different values for `config.long` and `config.lat`, see #495, so your current location might be wrong.

If that doesn't solve the issue, delete all rules created by Hue Essentials through the API.  Hopefully you did create a separate username for Hue essentials, so you can identify them by their `owner` attribute.  If not, you might find resource links created by Hue Essentials listing the rules per routine.  Then re-create the "from dusk till dawn" in Phoscon.

bruesche, 2020-04-09:
So I have to delete the virtual daylight Sensor from database? How can I do that? How can I set the settings again with the deCONZ API?

Can you give advice?

ebaauw, 2020-04-09:
> So I have to delete the virtual daylight Sensor from database?

No.

> How can I set the settings again with the deCONZ API?

See http://dresden-elektronik.github.io/deconz-rest-doc/ for the API documentation.  You need to do a PUT to `/sensors/#/config` with a body of `{"long": "...", "lat": "..."}`, substituting `#` with the id of the Daylight sensor, and `...` with your coordinates.

I prefer to use `ph` to interact with the API.  This little utility comes with Homebridge Hue, and handles the username for you.  You can also use `curl` or postman, which provides a graphical interface.

bruesche, 2020-04-09:
Seems to be working great. Homebridge now gives no error at startup. The parameters "sunsetoffset" and "sunriseoffset" can be used in the way "sunsetoffset": "-120"? Is this correct that this config would decrease the sunset time -2 hours?

ebaauw, 2020-04-09:
Yeah, I think so.  I always get confused about the sign...

bruesche, 2020-04-09:
The sign?

ebaauw, 2020-04-09:
-120 vs 120

bruesche, 2020-04-10:
Works as suggested.
Thanks a lot!

closed 2020-04-10 by bruesche

mgrn0, 2020-05-09:
> No.
>
>
> See http://dresden-elektronik.github.io/deconz-rest-doc/ for the API documentation. You need to do a PUT to `/sensors/#/config` with a body of `{"long": "...", "lat": "..."}`, substituting `#` [...]

I cannot find the properties for lat and long in the API documentation, is this by accident or were they removed? (I am having a similar problem and I want to rule out possible causes; I updated the config according to your comment and the sensor shows 'configured: true' and an offset, but no location data. Is this correct?

mgrn0, 2020-05-09:
but thanks to @bruesche and @ebaauw for this thread, my motion sensors did not work for the exact same reason, maybe phoscon app should highlight a non-working daylight sensor (or a bad configuration) that makes the 'sensor' not deliver correct values?

mgrn0, 2020-05-10:
I have a very weird situation regarding my sunrise/sunset times: I realized that my lights were dimming way too early; 

I specified the values as suggested by @ebaauw ; 
The local time of my deconz is showing the correct values, the timezone is correct, however the "lastupdated" value for the daylight sensor is 2 hours off, sunrise and sunset are WAY off (I am from Austria, the sunset today was around 20:25 but the API gives me for this sensor: 
> "sunrise": "2020-05-10T02:24:00",
> "sunset": "2020-05-10T15:07:16"

What can be the reason for that? I set my location through the api with the payload/body of
> {"long": "48.0", "lat": " 14.5"}
(simplified values for privacy's sake), are there any parameters I am missing of using in a wrong way?

ebaauw, 2020-05-10:
Times are in UTC.

mgrn0, 2020-05-11:
Ah, thanks ;) I was trying all different notations and just messed up the variables!