dresden-elektronik/deconz-rest-plugin #2674: Deconz some Legrand switch impossible to perform inclusion

x2005254, 2020-04-09:
Dear Deconz, Conbee 

Sorry First because I'm not developer at all, simply user, but IT engineering and i have some trouble with some news brand legrand switch and deconz, I succesfully added Simple wire &wireless switch, but the new Double wireless switch isn't reconize by Deconz  :
https://www.youtube.com/watch?v=Lp-x6LF7-Bc

you can find picture of the one not working on attachment
[image]

i had also problem to include this legrand netatmo gateway to use it on deconz as a device zigbee relay.
[image]

thanks in advance for consideration, help ! 

Finally, take care all of you with this worldwide healt crise COVID-19.

Smanar, 2020-04-09:
Hello, for the second question is not possible, this device is a complete Gateway, not a simple device.

For the first one I think nothing is needed, I have found all informations needed on another projet.
I can give you something to try tomorrow. Are you on Unix machine ? Do you know how to compile the code to make a try ?

x2005254, 2020-04-09:
Hello,

Thanks for reply!

Ok sade to me that gateway couldn’t be an normal device.

Anyway, for the first question, Please give me what I can do you to make it
work, deconz is actually running on jeedom, an domotic system front end
based on debian I guess. It’s linux for sure, with ssh access but you want
I can check distribution name if you need.

But I can added the device by deconz for Windows if it’s easier

Best regards

Smanar, 2020-04-10:
We can only compile the code for Unix, but I think it will work on your OS with SSh access.
So nothing dangerous for your system, theses command don't change your config.

You have all the procedure on mainpage https://github.com/dresden-elektronik/deconz-rest-plugin
>Install deCONZ development package (optional, Linux only)

Just at step 1 Use this url instead
git clone https://github.com/Smanar/deconz-rest-plugin.git

The compilation can take several minuts, around 5 mn for a raspberry 3.

And you can make a backup of the original plugin file /usr/share/deCONZ/plugins/libde_rest_plugin.so before updating the file.

You just need to close deconz, before updating the file, no danger for your setup, no setting change, and you can make rolllback the plugin file if you have made a backup.

I don't think the device will appear in Phoscon, but I think you will see it on jeedom.

x2005254, 2020-04-10:
Hi again,

Ok i will do it, but the link you send it to me there not installation
guide only all issue open and mine, no step to follow
https://github.com/dresden-elektronik/deconz-rest-plugin

Smanar, 2020-04-10:
Yes, it is, search the part that have as title "Install deCONZ development package (optional, Linux only)"

C'est sur la page d'acceuil du github, le paragraphe commence par "Install deCONZ development package (optional, Linux only)"

zinoalex, 2020-04-13:
Good morning guys, were you able to add it? I have a couple of those wireless switch (K4570CW and K4027CW) and few of the wired deviators K4003C (https://catalogo.bticino.it/BTI-K4003C-IT) by Bticino/Legrand but I'm unable to pair the device with my zigbee network. K4531C (power monitor outlet) and F20T60A (DIN power consumption module) are working great.

Smanar, 2020-04-13:
Hello
@zinoalex can you give me the model ID used by your device in deconz pls ? (On Basic cluster 0000, you probably need to wake up the device when pressing the "read" button)

ATM I m using "Double gangs remote switch" for him.

zinoalex, 2020-04-13:
Hi Smanar,
in the Model Identifier under Basic
K4531C: Connected outlet
F20T60A: DIN power consumption module
all the others are not recognized in the network, that's the problem

I commented on this post because is the same manufacturer and the double gangs remote switch has the same hardware of K4570CW and K4027CW.

Smanar, 2020-04-13:
Yep, I m asking for not working device ^^

K4570CWis the Home/away switch ? These one is special, long story, but not used.
K4027CW is the wireless shutter control ? It need to have "Shutters central remote switch" as model id
K4003C is the wired switch (an bticino are smaller than Legrand one ???) It need to have as model id "Dimmer switch w/o neutral"

Here it's a double gang with "Double gangs remote switch" as Model id

It's possible the wired switch have another model id (this one is 2 x smaller).
You can't see the device in phoscon, but can you see it in deconz ? At least the wired one will appear even witout full integration.

zinoalex, 2020-04-14:
No and that is the problem, I'm unable to pair the device, both wireless and wired, with deconz, only the smart outlet works

Smanar, 2020-04-14:
@zinoalex whitch one deconz version have you ?

For the wired one
You set phoscon in permit join (add new light and not switch)
Long press (8s, keep pressing after first blink), then short press, wait 5s, retry if you haven't the green led.

zinoalex, 2020-04-14:
I'm using the 2.05.75 and I'm looking both at phoscon and deconz but no success even with your process. I'm starting to think that it's just incompatible but it's strange since the others works...

Smanar, 2020-04-14:
Yep it's strange, someone have the device working here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2456#issue-565888334

The led never change it's color ? how it's blinking ?

zinoalex, 2020-04-14:
Problem, found, mine was at firmware 15, upgrade to firmware 22 and now it works. The only issue now is that since is a deviator only works in toggle mode, not ON/OFF so in phoscon/Home asssistant the state is not synced, some software to play around.... If you want to add those devices in the compatibility list now you can. Thanks Smanar!

Smanar, 2020-04-14:
NP, I did not know for the firmware.
But your device haven't 3 positions ? neutral / on / off by defaut ?

zinoalex, 2020-04-15:
It has 4 pins: L , N , 1 and 2. When you press the hardware switch the contact changes between pin 1 and 2, like any other deviator, but in phoscon or Home Assistant you can only use it in toggle mode, in deconz you can use only send the toggle and the ON command, if you send the off command nothing appens. This means that in HA and phoscon the status of pins is not synced.

Smanar, 2020-04-15:
Haaa, ok, so it's not the same device than legrand at all.
Legrand one juste have Line in / Line out. So it's not the same working mode, and it have dimmer feature.
There is so much difference, I have a doubt now about the "same firmware", you have the model id of the device from deconz ?

And now I understand better, there is no real on/off because you can put a bulb on 1 and 2 (It's stupid, but possible), and it s a toggle mode.

zinoalex, 2020-04-15:
It's actually quite a smart device because you can use it on a normal switch line, deviated line and on an inverted line without changing anything except the smart switch
[image]
in the model id i have: light switch with neutral
[image]
I was talking about this toggle function in deconz
[image]

x2005254, 2020-04-15:
Hi there,

Please find bellow my installation of SDK steps :
xXx@Jeedom-xXx:~$ cd DeconzSDK/
xXx@Jeedom-xXx:~/DeconzSDK$ sudo apt install deconz-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package deconz-dev
xXx@Jeedom-xXx:~/DeconzSDK$ git clone
https://github.com/Smanar/deconz-rest-plugin.git
Cloning into 'deconz-rest-plugin'...
remote: Enumerating objects: 30, done.
remote: Counting objects: 100% (30/30), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 10188 (delta 16), reused 17 (delta 7), pack-reused 10158
Receiving objects: 100% (10188/10188), 8.91 MiB | 9.48 MiB/s, done.
Resolving deltas: 100% (6913/6913), done.
xXx@Jeedom-xXx:~/DeconzSDK$ ll
total 12
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:16 .
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:16 ..
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:17 deconz-rest-plugin
xXx@Jeedom-xXx:~/DeconzSDK$  cd
xXx@Jeedom-xXx:~$ ll
total 44
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:16 .
drwxr-xr-x 4 root     root     4096 Jan  3 01:05 ..
-rw------- 1 xXx xXx  207 Apr 10 22:00 .bash_history
-rw-r--r-- 1 xXx xXx  220 Apr 18  2019 .bash_logout
-rw-r--r-- 1 xXx xXx 3526 Apr 18  2019 .bashrc
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:07 deconz-backup
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:16 DeconzSDK
drwx------ 3 xXx xXx 4096 Jan  3 14:31 .gnupg
-rw-r--r-- 1 xXx xXx  807 Apr 18  2019 .profile
drwx------ 2 xXx xXx 4096 Apr  9 15:55 .ssh
-rw------- 1 xXx xXx 3882 Apr  9 15:55 .viminfo
xXx@Jeedom-xXx:~$ ll
total 44
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:16 .
drwxr-xr-x 4 root     root     4096 Jan  3 01:05 ..
-rw------- 1 xXx xXx  207 Apr 10 22:00 .bash_history
-rw-r--r-- 1 xXx xXx  220 Apr 18  2019 .bash_logout
-rw-r--r-- 1 xXx xXx 3526 Apr 18  2019 .bashrc
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:07 deconz-backup
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:16 DeconzSDK
drwx------ 3 xXx xXx 4096 Jan  3 14:31 .gnupg
-rw-r--r-- 1 xXx xXx  807 Apr 18  2019 .profile
drwx------ 2 xXx xXx 4096 Apr  9 15:55 .ssh
-rw------- 1 xXx xXx 3882 Apr  9 15:55 .viminfo
xXx@Jeedom-xXx:~$ cd DeconzSDK/
xXx@Jeedom-xXx:~/DeconzSDK$ ll
total 12
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:16 .
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:16 ..
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:17 deconz-rest-plugin
xXx@Jeedom-xXx:~/DeconzSDK$ cd deconz-rest-plugin/
xXx@Jeedom-xXx:~/DeconzSDK/deconz-rest-plugin$ ll
total 2588
drwxr-xr-x 6 xXx xXx   4096 Apr 15 21:17 .
drwxr-xr-x 3 xXx xXx   4096 Apr 15 21:16 ..
-rw-r--r-- 1 xXx xXx   7617 Apr 15 21:17 authorisation.cpp
-rw-r--r-- 1 xXx xXx   5978 Apr 15 21:17 basic.cpp
-rw-r--r-- 1 xXx xXx 132079 Apr 15 21:17 bindings.cpp
-rw-r--r-- 1 xXx xXx   2975 Apr 15 21:17 bindings.h
...
...
-rw-r--r-- 1 xXx xXx  46791 Apr 15 21:17 zcl_tasks.cpp
xXx@Jeedom-xXx:~/DeconzSDK/deconz-rest-plugin$ cd ..
xXx@Jeedom-xXx:~/DeconzSDK$ ll
total 12
drwxr-xr-x 3 xXx xXx 4096 Apr 15 21:16 .
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:16 ..
drwxr-xr-x 6 xXx xXx 4096 Apr 15 21:17 deconz-rest-plugin
xXx@Jeedom-xXx:~/DeconzSDK$ cd deconz-rest-plugin
xXx@Jeedom-xXx:~/DeconzSDK/deconz-rest-plugin$ git checkout -b mybranch HEAD
Switched to a new branch 'mybranch'
xXx@Jeedom-xXx:~/DeconzSDK/deconz-rest-plugin$ qmake && make -j2
Info: creating stash file
/home/xXx/DeconzSDK/deconz-rest-plugin/.qmake.stash
Project *ERROR*: Unknown module(s) in QT: serialport websockets
xXx@Jeedom-xXx:~/DeconzSDK/deconz-rest-plugin$

>

x2005254, 2020-04-15:
do you have a precompiled version for debian :
xXx@Jeedom-xXx:~/DeconzSDK/deconz-rest-plugin$ uname -a
Linux Jeedom-xXx 4.19.0-8-arm64 #1 SMP Debian 4.19.98-1 (2020-01-26)
aarch64 GNU/Linux

Le mer. 15 avr. 2020 à 23:24, sofiane helou <sofiane.helou@gmail.com> a
écrit :

> Hi there,
>
> Please find bellow my installation of SDK steps :
> xXx@Jeedom-xXx:~$ cd DeconzSDK/
> xXx@Jeedom-xXx:~/DeconzSDK$ sudo apt install deconz-dev
> Reading package lists... Done
> Building dependency [...]

zinoalex, 2020-04-16:
> 
> 
> It's actually quite a smart device because you can use it on a normal switch line, deviated line and on an inverted line without changing anything except the smart switch
> [image]
> in the model id i have: light switch with neutral
> [image]
> I was talking about this toggle function in deconz
> [image]

Small update: shutter wireless control K4027CW added to the network and working fine with HA. You just need to switch the whole network on channel 11. Now I'm fighting with the deviator and the wired shutter control

Smanar, 2020-04-16:
@x2005254, there is precompiled version, I think you are using it ATM, but you need to compile it with the modification to test it, the pre-compiled version don't have the modification.
Or you are asking for deconz-dev instead of deconz ?

BTW:
>Unable to locate package deconz-dev

Try first 
`sudo apt update`

Else if you have debian 64bit I think ubuntu packages will work http://deconz.dresden-elektronik.de/ubuntu/beta/ for deconz-dev.

@zinoalex nice device, I did not know it. But How are you using it in HA, how are you using the toogle command with API ? BTW, this device is not yet in deconz, so perhaps you miss too state return (need to put it in whitelist in bind file)

x2005254, 2020-04-16:
I did it for sure, and I haven't the source list for dev package :
$ cat /etc/apt/sources.list.d/deconz.list
deb http://phoscon.de/apt/deconz buster main
deb http://phoscon.de/apt/deconz buster-beta main
deb http://phoscon.de/apt/deconz buster-dev main

i tried today after update some lib to compile again and iI had error on
make -j2, the qmake work actually, i'm not on raspberry but on freebox
delta with jeedom template based on debian.
please see bellow errors of make :

n$ qmake
Info: creating stash file
/home/x2005254/DeconzSDK/new/deconz-rest-plugin/.qmake.stash
xXx@Jeedom- xXx :~/DeconzSDK/new/deconz-rest-plugin$ make -j2
make -f Makefile.Release
make[1]: Entering directory '/home/ xXx /DeconzSDK/new/deconz-rest-plugin'
/usr/lib/qt5/bin/uic -no-stringliteral de_web_widget.ui -o
ui_de_web_widget.h
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W
-D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS
-DGW_SW_VERSION=\"2.05.75\" -DGW_SW_DATE=1587056713
-DGW_API_VERSION=\"1.16.0\"
-DGIT_COMMMIT=\"4fbd0ed2d68439b3651887dbdc9044661ec028b7\"
-DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500
-DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700
-DGW_MIN_AVR_FW_VERSION=0x26330500 -DGW_MIN_R21_FW_VERSION=0x26490700
-DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300
-DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB
-DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB
-DQT_CORE_LIB -I. -I../.. -I../../common -isystem
/usr/include/aarch64-linux-gnu/qt5 -isystem
/usr/include/aarch64-linux-gnu/qt5/QtWidgets -isystem
/usr/include/aarch64-linux-gnu/qt5/QtGui -isystem
/usr/include/aarch64-linux-gnu/qt5/QtSerialPort -isystem
/usr/include/aarch64-linux-gnu/qt5/QtWebSockets -isystem
/usr/include/aarch64-linux-gnu/qt5/QtNetwork -isystem
/usr/include/aarch64-linux-gnu/qt5/QtCore -Irelease -isystem
/usr/include/libdrm -I. -I/usr/lib/aarch64-linux-gnu/qt5/mkspecs/linux-g++
-o release/authorisation.o authorisation.cpp
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W
-D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS
-DGW_SW_VERSION=\"2.05.75\" -DGW_SW_DATE=1587056713
-DGW_API_VERSION=\"1.16.0\"
-DGIT_COMMMIT=\"4fbd0ed2d68439b3651887dbdc9044661ec028b7\"
-DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500
-DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700
-DGW_MIN_AVR_FW_VERSION=0x26330500 -DGW_MIN_R21_FW_VERSION=0x26490700
-DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300
-DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB
-DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB
-DQT_CORE_LIB -I. -I../.. -I../../common -isystem
/usr/include/aarch64-linux-gnu/qt5 -isystem
/usr/include/aarch64-linux-gnu/qt5/QtWidgets -isystem
/usr/include/aarch64-linux-gnu/qt5/QtGui -isystem
/usr/include/aarch64-linux-gnu/qt5/QtSerialPort -isystem
/usr/include/aarch64-linux-gnu/qt5/QtWebSockets -isystem
/usr/include/aarch64-linux-gnu/qt5/QtNetwork -isystem
/usr/include/aarch64-linux-gnu/qt5/QtCore -Irelease -isystem
/usr/include/libdrm -I. -I/usr/lib/aarch64-linux-gnu/qt5/mkspecs/linux-g++
-o release/bindings.o bindings.cpp
In file included from bindings.cpp:11:

*de_web_plugin.h:16:10: fatal error: deconz.h: No such file or
directory #include <deconz.h>          ^~~~~~~~~~compilation
terminated.make[1]: *** [Makefile.Release:521: release/bindings.o] Error
1make[1]: *** Waiting for unfinished jobs....In file included from
authorisation.cpp:13:de_web_plugin_private.h:23:10: fatal error: sqlite3.h:
No such file or directory #include <sqlite3.h>
^~~~~~~~~~~compilation terminated.make[1]: *** [Makefile.Release:503:
release/authorisation.o] Error 1make[1]: Leaving directory
'/home/xXx/DeconzSDK/new/deconz-rest-plugin'make: *** [Makefile:40:
release] Error 2*

x2005254, 2020-04-16:
and I'm in lasted stable version :
https://github.com/dresden-elektronik/deconz-rest-plugin/releases

do you think you will add these legrand shutter soon on next stable
release?

Smanar, 2020-04-17:
OK, lol I have checked my version 
>deconz-dev is already the newest version (2.05.44)

So now I understand why I have some problem too during compilation.

So to update (Realy out of date) my version I have used

```
wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -
sudo sh -c "echo 'deb http://phoscon.de/apt/deconz $(lsb_release -cs)-beta main' > /etc/apt/sources.list.d/deconz.list"
sudo apt update
sudo apt upgrade
```

And now
>deconz-dev is already the newest version (2.05.75-raspbian-stretch-beta).

x2005254, 2020-04-17:
Hi,
Still impossible to install for deconz-dev

```
$ sudo dpkg -i deconz-dev-2.05.75.deb
dpkg: error processing archive deconz-dev-2.05.75.deb (--install):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 deconz-dev-2.05.75.deb

```

there is a deconz-dev package for arm64 ??

Smanar, 2020-04-18:
Hello, have you tried commands from my previous post ?

x2005254, 2020-04-18:
Did you see my code section log on my message, I tried 2 thing.

Manual deb package with dpkg command.

And source install with update key apt update and apt upgrade and apt install deconz-dev nothing works.

Which post do you talk about?

Smanar, 2020-04-18:
>And source install with update key apt update and apt upgrade and apt install deconz-dev nothing works.

Was this one. You had some errors messages ?

x2005254, 2020-04-20:
Hi again,

I did the whole command you send and only one erroro at the end for sudo apt install deconz-dev.

Anywaty I will show you again :
- 1st command ok 
`$ wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -

--2020-04-20 10:35:20--  http://phoscon.de/apt/deconz.pub.key
Resolving phoscon.de (phoscon.de)... 144.76.96.194
Connecting to phoscon.de (phoscon.de)|144.76.96.194|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1692 (1.7K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                                 100%[============================================================================================================>]   1.65K  --.-KB/s    in 0s

2020-04-20 10:35:20 (92.3 MB/s) - written to stdout [1692/1692]

OK
`

- 2nd command  OK 
`$ sudo sh -c "echo 'deb http://phoscon.de/apt/deconz $(lsb_release -cs)-beta main' > /etc/apt/sources.list.d/deconz.list"
$ cat /etc/apt/sources.list.d/deconz.list
deb http://phoscon.de/apt/deconz buster-beta main
`

- 3rd command ok :
`$ sudo apt update
Hit:1 http://deb.debian.org/debian buster InRelease
Get:2 http://deb.debian.org/debian buster-updates InRelease [49.3 kB]
Get:3 http://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Hit:4 http://security.debian.org buster/updates InRelease
Hit:5 http://phoscon.de/apt/deconz buster-beta InRelease
Hit:6 https://deb.nodesource.com/node_12.x buster InRelease
Get:7 http://deb.debian.org/debian buster-backports/main Sources.diff/Index [27.8 kB]
Get:8 http://deb.debian.org/debian buster-backports/main arm64 Packages.diff/Index [27.8 kB]
Get:9 http://deb.debian.org/debian buster-backports/main Sources 2020-04-20-0804.50.pdiff [791 B]
Get:9 http://deb.debian.org/debian buster-backports/main Sources 2020-04-20-0804.50.pdiff [791 B]
Get:10 http://deb.debian.org/debian buster-backports/main arm64 Packages 2020-04-20-0804.50.pdiff [508 B]
Get:10 http://deb.debian.org/debian buster-backports/main arm64 Packages 2020-04-20-0804.50.pdiff [508 B]
Fetched 153 kB in 2s (67.0 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
`

- 4st command ok 
`$ sudo apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  python-bcrypt python-blinker python-fasteners python-jwt python-lockfile python-monotonic python-nacl python-oauthlib python-paramiko python-pexpect python-ptyprocess python-pyasn1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

- Last command KO 
`$ sudo apt install deconz-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
$ sudo apt install deconz-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package deconz-dev
`

this is the only errors I met "**E: Unable to locate package deconz-dev**"

regarding the message I don't think that there is deconz-dev package available for amr64 architecture in the sources list you gave to me.

Thanks in advance for your help

Smanar, 2020-04-20:
Idk, I have see a special folder for arm64, but file aren't called deconz-dev, you are probably right.

I realy don't have idea, was my last chance command. I can submit the pull as "untested", but you will need to come back after for button return test, and if it don't work wait the next again version ...

x2005254, 2020-04-20:
Thanks for reply.

So I will wait one of them (Button or version).

May I open a request on https://github.com/dresden-elektronik/deconz-rest-plugin/pulls, to wish that my 2 legrand devices to be included on next deconz version ?

To remember I have the :
- 2 simple legrand wireless Switch 
https://www.legrand.fr/pro/catalogue/43070-version-celiane-with-netatmo/commande-double-sans-fil-celiane-with-netatmo-pour-eclairage-ou-prise-connectee-ou-micromodule-titane

- Arrival/Departure Switch
https://www.legrand.fr/pro/catalogue/42572-version-celiane-with-netatmo/commande-generale-sans-fil-departarrivee-pour-installation-connectee-celiane-with-netatmo-titane

Smanar, 2020-04-20:
NP, I will make the PR myself tomorrow (I have some other pr to do, and all are the same fork), I think it will be validated even we can't test it. So except if the V 76 was already ready somewhere you will have at next version, but untested not sure it will work.

That is for the double legrand wireless

For the arrival/departure you have the official isssue here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883
But to resume, this device can work but
-    You need to change the MAC adress of the gateway
-    It seeem there is a difference if the switch is sell with the gateway or alone.
-    If you make bad actions, you can make your Legrand geteway unusable.

Smanar, 2020-04-21:
You have reset the gateway and the remote in same time ?

The procedure here
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-586609284

You need to change the conbee MAC adress (using deconz), to mimic the same MAC adress of the legrand gateway.
You will have the adress printed on devices, there is 2 adress on the gateway, but only 1 similar at the one in the remote.

But take care, I have never do that on my side, but I can understand if you loose all your devices, because I think devices will use it too, to reconize their gateway.
**It's realy not a safe procedure, better avoid it if you haven't an empty network**.

x2005254, 2020-04-21:
Thanks again Smanar.

I can understand to change my conbee mac address now is not recommanded, as all my network working well with many device, and I don't want to redo again.

and the big issue that isn't usable as other with normal code on jeedom and deconz plugin need special code called websocket for one remote is too tricky for me.

I keep it on my mind if one day have second conbee implemented.

x2005254, 2020-04-28:
Hi @Smanar, 

I have one question about all legrand switch on phoscon, I have note the same configuration possible with Philips switches and Legrand Switches :
Philips create a group switch when you can add ligth on it automatically after inclusion, and this permit to make group binding between switch and devices :
[image]

But for all Legrand Switches I don't have this luck, it's see like a other device, within I can't do nothing on phoscon with :
[image]

I tried to make it by Deconz Gui and Bind unatary, but I had some random On/Off failed when I want to turn on or off the light sometime work sometime not, also after many try.

I remark that Legrand make automatic affect binding of the closer device, because when I had new legrand switch included (without bindigng set), i can go anywhere on my home and it will operate with the closest device, when I push on or off, is it for that reason, my binding unatary set on deconz work ramdomly?

Should I make a request to make legrand switch creating group automatically after inclusion? on next release?

Smanar, 2020-04-28:
>I remark that Legrand make automatic affect binding of the closer device, because when I had new legrand switch included (without bindigng set), i can go anywhere on my home and it will operate with the closest device, when I push on or off.

You mean when you include a new wireless switch in deconz, It control the closest light during the pairing ? Or if you move It control the closest light every time ? And without setting nothing on group ?

Legrand device have group too, but they are not displayed in Phoscon (and I can do nothing for that, phoscon is closed source), We can ask for support, but looking for last official update, I think they have bigger problems ATM.
Using the GUI is hard because the group gestion is not easy in it, I think it will be easier using API https://dresden-elektronik.github.io/deconz-rest-doc/groups/
To identify the specific device group check the "devicemembership" field.
BTW , have you tried with "Open Wireless Light Control (2016)", the Legrand group is visible and manageable in it.

On the other Legrand topic, there is someone that know well how to use legrand group, @smesguich if I m right.

But why do you want  to use this feature ? Your home automation application can make scenario with existing switch ?

smesguich, 2020-04-28:
I confirm, groups are visible via Open Wireless Light Control (2016) app. You need to go in Phoscon in "Help -> Old WebApp version : Open Wireless Light Control (2016)"
[image]

@Smanar  the advandage of this feature is the low-level feature, very fast regarding any scenario in any home automation system.

Smanar, 2020-04-28:
Yes, but as it is not managed by the home automation application you miss
- Logs, statistic
- Information to be used for other scenario

x2005254, 2020-04-28:
> You mean when you include a new wireless switch in deconz, It control the closest light during the pairing ? Or if you move It control the closest light every time ? And without setting nothing on group ?

It look like assign after inclusion, but randomly, and sometime I i change the place of switch during hours and previous meshing is lost, it affect a new binding light group, I'm so supprised of this weird usage.

> Legrand device have group too, but they are not displayed in Phoscon (and I can do nothing for that, phoscon is closed source), We can ask for support, but looking for last official update, I think they have bigger problems ATM.

Thanks for sharing information, I wasn't aware that Phoscon is different source.

> Using the GUI is hard because the group gestion is not easy in it, I think it will be easier using API https://dresden-elektronik.github.io/deconz-rest-doc/groups/

I'm fully agree that isn't easy.

> To identify the specific device group check the "devicemembership" field.
> BTW , have you tried with "Open Wireless Light Control (2016)", the Legrand group is visible and manageable in it.
> On the other Legrand topic, there is someone that know well how to use legrand group, @smesguich if I m right.

I'm going to study this way also.

> But why do you want to use this feature ? Your home automation application can make scenario with existing switch ?

Sure my Jeedom home control work well, but I need security in case Jeedom is out of work, sometimes, where you not home, home automation crashe then I receive bad feedback of family (wife & children) that nothing work anymore, so for simple think like light I don't want to be locked by jeedom himself :D

x2005254, 2020-04-28:
Thanks again @Smanar it solved with managing group with API, many more better than binding 👍 

`{
"action": {
"bri": 127,
"colormode": "hs",
"ct": 0,
"effect": "none",
"hue": 0,
"on": false,
"sat": 127,
"scene": null,
"xy": [
  0,
  0
],
},
"devicemembership": [
  "22"
],
"etag": "7cf5dddd48ae2ff80a1720e1060e375f",
"id": "16",
"lights": [
  "7",
  "10"
],
"name": "Remote switch 22",
"scenes": [],
"state": {
"all_on": false,
"any_on": false
},
"type": "LightGroup",
"uniqueid": "00:04:74:00:00:ac:d1:8e"
}`

Smanar, 2020-05-01:
@x2005254 Ok so good news, the double gang is working https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-622359415

But IDK when we will have the next version.

x2005254, 2020-05-01:
> is working #883 (comment)

Sound great !! Thanks

x2005254, 2020-05-28:
Hi there,

I updated deconz 2.05.77 I tried to include a wireless dual gang because you told me it would be managed at the next update.

So via Phoscon, the inclusion gave nothing, but via the deconz plugin under jeedom, it worked:

only problem only one of the two event buttons work (in terms of change of value when I click on the physical button):

- the first rermonte 1002 or 2002 in function if I press on or off with my left button.

- but the second still displays 3002 regardless of the impulson made on the right button of the inter.

do you think this is an API problem? of inclusion? partial compatibility?

Smanar, 2020-05-28:
Oups, sorry I m looking the code, and seeing the problem, there is, a typo on a command. Sorry, but I think the future update will be faster.

For inclusion, it's just luck, using phoscon or jeedom do exactly the same thing.

I don't remember, you have updated your device ? Because on updated one you have more command and there is only one broken.

Smanar, 2020-05-29:
@x2005254 if you haven't the gateway to update your device and wana test the others command (like long press) PM me on deconz,
I have the username but I m rarely online (using the browser app)

Smanar, 2020-06-24:
@x2005254 if you can test the versio 78, the typo bug is corrected.
If it s still not working, I will need somes informations.

x2005254, 2020-06-24:
Thanks @smanar

This week I’m in business trip but next week I will be aboe to test.

Thanks a lot

closed 2020-07-25 by stale[bot]

lambu76, 2020-11-04:
Hi guys,
I tested a wireless switch Wake up/Sleep (L4574CW) with deConz v.  2.05.86.

It has been properly recognized and added into zigbee network and after accessing Phoscon old web app I was able to view the preconfigured group "Remote switch Wake up / Sleep 3".
After assigning some lights I can switch them ON and OFF all together ONLY by using the webinterface, but if I push the physical button nothing happens.

Someone could help me ?

thx

samsam-rolon, 2020-11-04:
> Hi guys,
> I tested a wireless switch Wake up/Sleep (L4574CW) with deConz v. 2.05.86.
>
> It has been properly recognized and added into zigbee network and after accessing Phoscon old web app I was able [...]

I didn't test it yet (event if I own it, it's still connected on the official Legrand GW) but as I can imagine, it act as a "scene button" and not as a classical switch ... so it may need to associate some scene to some light. Because the device is not calling an "On/Off" action

lambu76, 2020-11-04:
it has a scene assigned but it seems nothing happens at button press neither for WakeUp or Sleep.

Could you please make a try with your one ?

Smanar, 2020-11-04:
> It has been properly recognized and added into zigbee network and after accessing Phoscon old web app I was able to view the preconfigured group "Remote switch Wake up / Sleep 3".
> After assigning [...]

He realy ?
It the one sell with the gateway, or you have buy it alone ? It s complicated normally to include it, because you need to set a MAC adress (at least for first type)

From my memory if you look at the websocket log, every time you use it, you will see a "scene" message.

lambu76, 2020-11-04:
BTicino sold it alone without gateway....at least in Italy in this moment

samsam-rolon, 2020-11-04:
> 
> 
> it has a scene assigned but it seems nothing happens at button press neither for WakeUp or Sleep.
> 
> Could you please make a try with your one ?

I will, probably next week during my holidays

> 
> 
> > It has been properly recognized and added into zigbee network and after accessing Phoscon old web app I was able to view the preconfigured group "Remote switch Wake up / Sleep 3".
> > After assigning some lights I can switch them ON and OFF all together ONLY by using the webinterface, but if I push the physical button nothing happens.
> 
> Ha realy ?
> It the one sell with the gateway, or you have buy it alone ? It s complicated normally to include it, because you need to set a MAC adress (at least for first type)
> 
> From my memory if you look at the websocket log, every time you use it, you will see a "scene" message.
> 
> But this device was never integrated, because of those constraints.

The Day/Night is always sell alone. It's the Leave/Comming Home which is sold with the official GW

Smanar, 2020-11-04:
>The Day/Night is always sell alone. It's the Leave/Comming Home which is sold with the official GW

Ha yep, I m making mistake. But It s worse there is absolutly nothing for this device in deconz yet ^^.

@lambu76 If you can see it in the API, can you share the device JSON.

I m interested too by the device description from deconz (basic attribute and cluster list).

I think it will be easier to trigger a button event from the remote than using the scene feature from the device.

lambu76, 2020-11-04:
I will be more than happy to help you, but I need your help to understand how to provide you the information requested.

How to grab JSON datas from the device ?

Smanar, 2020-11-04:
Wich one third app are you using ?

lambu76, 2020-11-04:
Starting from your first url, I'm getting this error message

http://192.168.0.172:81/api/API_KEY/sensors

`[{"error":{"address":"/sensors","description":"unauthorized user","type":1}}]`

Smanar, 2020-11-04:
You need to replace "API_KEY" by your key.
You can use the same one than in your third app, or use the last command to generate one.

lambu76, 2020-11-04:
got it....I'm using Home Assistant

[10 more replies]