dresden-elektronik/deconz-rest-plugin #2676: Conbee II USB stick on PI4: can't get API KEY

vdiogo, 2020-04-09:
I have a Conbee II USB stick on a PI4 running the deCONZ addon on HASSIO, and I need to get the API KEY so I can connect deCONZ to node-red.

After getting the IP and port using the instructions here https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/, the POST request to get the API KEY is not successful (ends with a _Could not get any response_ error after running for about 1 min).

Any suggestion on what could be the issue?

Smanar, 2020-04-10:
For me it's not a API key problem.
You are sure for ip and port ? If you use them on a web browser you will obtain the old web app from deconz ?

vdiogo, 2020-04-10:
> For me it's not a API key problem.
> 
> You are sure for ip and port ? If you use them on a web browser you will obtain the old web app from deconz ?
> 
> 
> 
> Else it will be the same IP than you are using for Phoscon.

Can you elaborate on your comment? What do you mean when you say it’s not an API key problem? And how should I test the IP on a web browser?

Smanar, 2020-04-11:
If it was API problem, you will have an error message like "not autorised" or a bad answer. Here you have a connexion problem.

Just type in a browser http://IP:PORT then press enter, if you are right you will have the  old webapp, else you are using bad ip or have something blocking in your network.

CarpeDiemRo, 2020-04-28:
I have the same issue. I'm using Postman on macOS. 

at 
`GET https://dresden-light.appspot.com/discover `

 I receive the 200 OK, but at 

`POST 172.30.33.1:40850/api`
or 
`POST https://172.30.33.1:40850/api`

I'm receiving 

> Could not get any response
There was an error connecting to http://172.30.33.1:40850/api.
Why this might have happened:
The server couldn't send a response:
Ensure that the backend is working properly
Self-signed SSL certificates are being blocked:
Fix this by turning off 'SSL certificate verification' in Settings > General
Proxy configured incorrectly
Ensure that proxy is configured correctly in Settings > Proxy
Request timeout:
Change request timeout in Settings > General

The body content is 
```
{
    "devicetype": "myapplication"
}
```
raw button checked

Any advice or suggestions?

mmychu, 2020-04-30:
I have the same problem with getting API key
After about 1 minute, there is an error after POST request
[image]

mmychu, 2020-04-30:
Set the correct PORT, restart decozn and try again
[image]

CarpeDiemRo, 2020-05-10:
Thanks for your answer. It Works!

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix