dresden-elektronik/deconz-rest-plugin #2677: Device Support Tuya Smart Plug TS0121

bilbon0bud, 2020-04-10:
Hello

here is a smart plug zigbee from tuya, she have a power meter report, but i can't see him in deconz or in home assistant.
This device isn't inside the supported device list, so i give you some elements to integrate it in deconz.

Thanks for Your work.

Best Regards

[image]
[image]
[image]
[image]

SwoopX, 2020-04-10:
I can take care of that. Can you please also provide screenshots from the Electrical Measurement cluster as well (make sure the attributes have been read)? Thanks!

bilbon0bud, 2020-04-10:
[image]

do you need more ?

bilbon0bud, 2020-04-10:
[image]

SwoopX, 2020-04-10:
Thanks. Can you please apply a load on the plug and take another screenshot (1st one of the 2)? Since neither multipliers nor divisors are maintained, I need to know the active power to get the units right.

bilbon0bud, 2020-04-10:
[image]

bilbon0bud, 2020-04-10:
here is the cature with a charge.

SwoopX, 2020-04-10:
Thanks again. How much power was it? I'd assume 96W?

bilbon0bud, 2020-04-10:
I ll try tomorrow the same test with a zwave plug from fibaro to give you the answer.

I plug my fruit juicer so 96w seem to be a good response.

Thanks for tour work see you tomorrow

SwoopX, 2020-04-10:
Cool. And don't forget to also make a screenshot of the basic cluster while the attributes have been read. Pretty sure I already know what's in there, but I wanna be on the safe side.

bilbon0bud, 2020-04-11:
[image]
[image]
[image]

the power you gave me seem to be good.

SwoopX, 2020-04-11:
Thanks! So in your last screenshot, the values already are as they should without any additional modification. Units are V, mA and W, great.

However, the Current Summation does not seem to fit your last screenshot (from the app). Deconz is saying 61 (presumably Wh) and the app much more. Was the device reset recently?

bilbon0bud, 2020-04-11:
Hello

The device as been unplugged.
I don’t reset it, but i don’t use it too.

It was just plug for the test.

Best regards

Klagio, 2020-05-10:
Hi, do you know if its this model?

https://it.aliexpress.com/item/4000688486196.html?spm=a2g0o.productlist.0.0.7cbb1c206TOJki&algo_pvid=c1e518a8-418b-49ce-9255-8879d8eaf6a2&algo_expid=c1e518a8-418b-49ce-9255-8879d8eaf6a2-0&btsid=0ab6f8ad15891440488305543eff36&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

vdbrink, 2020-05-11:
@bilbon0bud I received today the same but when I bought it it was advertised as metered but they changed that now to no-metered (they removed that text).
Did you test if it's metered?
It doesn't send metered data over zigbee.

bilbon0bud, 2020-05-11:
It seem to be metered, i m waiting the next release of deconz to test the
plugs.
the change made by swoopX seem to be merged, now i just have to wait a
little more.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

mhin76, 2021-03-22:
Hi all. I also bought such a plug recently (Model ID TS0121 and Manufacturer Name _TZ3000_rdtixbnu). I see under deCONZ the RMS current and active power incl. values. But I don't see any power nor consumption info under Homebridge - Homebridge Hue plugin. Not sure if maybe there an update is missing. With what REST API call can I try it out directly on deconz to see if that part is correct?