dresden-elektronik/deconz-rest-plugin #2678: [Device support request] innr SP220 smart plug

SwoopX, 2020-04-11:
[image]

**Basic cluster**
[image]
[image]

**Level Control**
[image]

SwoopX, 2020-04-11:
Manufacturer specific luster **FC82**

```
ZigBee Cluster Library Frame, Mfr: INNR Lighting BV (0x1166), Command: Discover Attributes Response, Seq: 208
    Frame Control Field: Profile-wide (0x1c)
    Manufacturer Code: INNR Lighting BV (0x1166)
    Sequence Number: 208
    Command: Discover Attributes Response (0x0d)
    Attribute Status Record
        Attribute: 0x0006
        Data Type: Character String (0x42)
    Attribute Status Record
        Attribute: 0x0010
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0x0011
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0x0012
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0x0013
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0x0014
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0x0015
        Data Type: 32-Bit Unsigned Integer (0x23)
    Attribute Status Record
        Attribute: 0xfffd
        Data Type: 16-Bit Unsigned Integer (0x21)

```

ebaauw, 2020-04-11:
So indeed Zigbee 3.0 as expected and no _Simple Metering_ nor _Electrical Measurement_ as expected/feared.  What's the _Image Type_ (in the OTAU panel)?  The 0xFC82 cluster is new.  Do the values change, as load increases?  I do have an unused innr bridge lying around, if you want to sniff how it interacts with the SP 220.  However, last time I checked, it was really unsophisticated.

The device should already be exposed by the REST API plugin (as not-so-smart plug).  Only thing to check is the whitelisting in `bindings.cpp` for the attribute reporting.  Best change `modelId.startsWith(QLatin1String("SP 120"))` to `modelId.startsWith(QLatin1String("SP "))` (only in that file).

BTW how do you do to send the _Discover Attributes_?  deconz-cli-plugin?  What's the byte code?

kasteleman, 2020-04-11:
The SP 220 does Indeed not support Simple Metering nor Electrical
Measurement as confirmed by INNR itself!

So no need to snif because it only does on/off

SwoopX, 2020-04-11:
@ebaauw 

> What's the Image Type (in the OTAU panel)?

Image type is 0x0310

> Do the values change, as load increases?

No, none of the values changes a bloody bit, even after 1h of my Sonos plaing stuff. Looks pretty useless.

> BTW how do you do to send the Discover Attributes? deconz-cli-plugin?

Indeed. Pretty handy, at least for mains powered devices.

```
zclattrmanu 0xcaca 1 0xfc82 0x1166 0C00001D
               |   |    |      |    |   | |------- # of attributes to read (1D)
               |   |    |      |    |   |--------- # Attribute to start with (0000)
               |   |    |      |    |------------- Command (Discover Attributes)
               |   |    |      |------------------ Manufacturer code
               |   |    |------------------------- Manufacturer specific cluster
               |   |------------------------------ Endpoint
               |---------------------------------- Device
```

ebaauw, 2020-04-13:
Holy crap, while documenting the GUI's keyboard shortcuts, I discovered that you can also send this request from the GUI.  Select a cluster, select the node, press `9`.  Updated the "deCONZ for Dummies" Wiki.  It doesn't set the _Manufacturer Code_ though, so useless for clusters with that code.

SwoopX, 2020-04-15:
Gee, where did you get that from, trying?

ebaauw, 2020-04-15:
While editing the "deCONZ for Dummies" Wiki, I was double-checking the keys with the sniffer running.  I did try all digit keys, but not all the letters, there could be another hidden gem (I was hoping for a _Read Attributes_ of the _Basic_ cluster).

SwoopX, 2020-04-15:
Nice one.

ualex73, 2020-04-21:
No measurements is a "downgrade" from SP120, because I bought the SP120 for that ;-) So I can skip the SP220 completely.

kasteleman, 2020-04-21:
I mentioned that to INNR also. But the main customers use the HUE bridge for example and then there is no measurement available even with the SP120. Therefore they did not implement it in the SP220 and thereby reduced the costs for the new plug.

ualex73, 2020-04-21:
I understand the business logic behind it, but they just lost me as a customer for plugs ;-)

kasteleman, 2020-04-21:
And again, did mention that to them. 1 of the + of the SP120 and reason some people use that plug.
Would have been a nice upgrade because of ZigBee 3.0 and attribute reporting…...

SwoopX, 2020-04-21:
Indeed. From my perspective, power measurement was more a unique selling proposition. Since it's not in anymore, I'm out. Really enjoyed the SP 120.

The SP 220 are more expensive btw ;)

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

P1Rebo, 2021-11-27:
Unfortunately I can't pair with my Conbee I stick plugged in... Any suggestions? My SP 120 could be connected without any problems... thanks in advance!

jee67, 2022-08-19:
Anyone, is the Sp220 accessable  in the Phoscon app, and if so how should I do this?

lryjnks, 2022-09-10:
> Anyone, is the Sp220 accessable in the Phoscon app, and if so how should I do this?

Same question here..

lryjnks, 2022-09-10:
Ok, strangely enough I now see the innr in the menu Lights. But in the search it had not found anything.

[image]